LibrePCB/LibrePCB #586: Schematic Editor : pin numbering on schematic

doom-fr, 2019-11-18:
##### SUMMARY
When you route a circuit, some devices  has many pins with the same functionality (for example diode  or resistor network). Changing connections in such device help the routing without changing anything else.
To help this, it will be easier to have pin number on schematic.

##### SOLUTION
Add pin number on schematic when device is set (it can be just a tooltip)

ouabache, 2019-11-18:
This falls into the question of how do we package multiple units into a common part. When a designer places a resistor on a schematic they should not have to worry about whether it is a single part or in a resister pac array. The board designer will make that decision at a later date. The board designer will also swap units around to achieve the best routing. 

Back annotation only works when there is a 1:1 relation between the schematic and board. If you want to draw one schematic for use on multiple boards (which we do want to do). Then you cannot do back annotation, you have to do forward annotation.

doom-fr, 2019-11-18:
What is the use case of doing multiple boards with only one schematic ?

We can also do a forward annotated schematic (read only & different from schematic)
schematic -> board -> annotated schematic

And when you debug your board, it will be useful to know on which pad you have to put your oscilloscope.

Maybe just solving the issue https://github.com/LibrePCB/LibrePCB/issues/580 will be enough to do this.