LibrePCB/LibrePCB #587: Board Editor : impossible to select copper wire

doom-fr, 2019-11-18:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     bd616e14d1366486cf53cad8b8c7e3e67119d960
Build Date:       2019-11-17T15:33:39
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
sometimes, it is not possible to select copper wire

##### STEPS TO REPRODUCE
It happens, but I do not know in which exact conditions.
Sometimes it works, sometimes not.

1. click on copper wire (top)

##### EXPECTED RESULTS
it is selected

##### ACTUAL RESULTS
sometimes, it is not selected

##### ADDITIONAL INFORMATION
none

5n8ke, 2020-08-07:
Does it happen, when you click on a net junction?
NetPoints don't seem to be highlighted properly when selected.

ubruhin, 2023-02-21:
I suspect this issue has been fixed "automatically" in the mean time as there were a lot of improvements made in the board editor selection behavior after this issue has been opened. So let's close the issue now.

closed 2023-02-21 by ubruhin