LibrePCB/LibrePCB #588: Board Editor : sometimes hairs of same net disappear

doom-fr, 2019-11-18:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.3-unstable
Git Revision:     bd616e14d1366486cf53cad8b8c7e3e67119d960
Build Date:       2019-11-17T15:33:39
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
Sometimes when moving components, hairs of same net disappear.

##### STEPS TO REPRODUCE
Seams to be when two consecutive hairs of same net are aligned or quite aligned.
Not always reproductible.

[image]

Hairs are OK. And move a little bit and ...

[image]

##### EXPECTED RESULTS
hairs are drawn

##### ACTUAL RESULTS
Sometimes, hairs are not drawn.

##### ADDITIONAL INFORMATION

doom-fr, 2019-11-22:
Can you qualify this as a bug ?
It happens a lot.

doom-fr, 2019-11-24:
@ubruhin Can you have a look at this ?
It is really annoying, because when you align devices, the air wire disappears.

ubruhin, 2019-11-24:
Yeah I think I can try to fix this.

doom-fr, 2019-11-24:
Great !

doom-fr, 2019-11-30:
[image]

doom-fr, 2019-11-30:
[image]

doom-fr, 2019-12-02:
Just see that it also has a impact on DRC.
When the airwire is not drawn, the missing connexion error is not raised.

This is really a problem as it is so possible to forget to link some components.

Can you please fix it soon ??

ubruhin, 2019-12-02:
Yes yes, I added it to the 0.1.4 milestone ;)

doom-fr, 2019-12-02:
Sorry to be so impatient :-)

By this, I just want to say that now, it is not just a problem of UI but a problem with the DRC.
So, much more important.

ubruhin, 2020-04-22:
Fixed in #687.

closed 2020-04-22 by ubruhin