LibrePCB/LibrePCB #589: Library and Board Editor : Polygon can't be selected in its whole area

doom-fr, 2019-11-19:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.3-unstable
Git Revision:     bd616e14d1366486cf53cad8b8c7e3e67119d960
Build Date:       2019-11-17T15:33:39
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
Polygon can't be selected in its whole area.

##### STEPS TO REPRODUCE

1. draw a top copper area (filled) of 0.2mmx0.4mm
2. click the center and try to move => does not work
3. double-click the center => does not work
4. click on the top (last 0.1mm) and try to move => works !
5. double-click on top (last 0.1mm) => works !

Works also on bottom, but not on left and right.
If rotate it, it's only work on left and right edge.

If the polygon is bigger, it works in the center, except on edge (0.1mm around) but also on corner (0.1x0.1mm on each corner).

Seems to be the same with other layers.

##### EXPECTED RESULTS
Selection works on the whole area of the polygon.

##### ACTUAL RESULTS
Selection is only possible in part of the area

##### ADDITIONAL INFORMATION
no

ubruhin, 2022-09-15:
I tried to reproduce this, but didn't see any problem. If you still see this problem with the latest release, screenshots/videos would be very helpful. Otherwise I'll close this issue...

ubruhin, 2023-02-21:
As I was not able to reproduce the problem, and there have been many related improvements implemented in the mean time, I'll close this issue now. Feel free to reopen if you still experience this problem.

closed 2023-02-21 by ubruhin