LibrePCB/LibrePCB #595: Board Editor : Alignment & distribution tools

doom-fr, 2019-11-24:
##### SUMMARY
To help making wonderful boards, it will be useful to have alignment and distribution tools for devices (at least).

##### SOLUTION
implement alignment & distribution tools set of Inkscape

[image]

EdizonTN, 2019-11-24:
Yep, this will be helpfull!

> To help making wonderful boards, it will be useful to have alignment and distribution tools for devices (at least).

small correction:

To help making wonderful boards, it will be useful to have alignment and distribution tools for **selected** devices (at least).

dbrgn, 2019-11-24:
Yep, I also like the tool provided by Inkscape. It's quite flexible.

doom-fr, 2019-11-27:
It can be aligning devices, but also pad of devices.