LibrePCB/LibrePCB #598: Library editor : No copper area

doom-fr, 2019-11-25:
##### SUMMARY
Some devices need to have "no copper" area.
[image]

[image]

The problem is that when you have a plane, it is not possible to specify this

##### SOLUTIONS
Add a "No copper area" layer in Library editor and Board editor.

doom-fr, 2019-11-26:
I have added some precision in the solution paragraph above.

dbrgn, 2019-11-26:
I think those zones are often called "Keepout Zone", right? It should not only affect the planes, but also result in DRC warnings if traces are crossing that area.

doom-fr, 2019-11-26:
This is OK for me.
I just do not use the term "Keepout" as it is not 100% clear.
As sometimes you are not authorized to put component but you can put traces.
And sometimes, you cannot put silkscreen.

"No copper" is clear. But it can be "Top copper keepout area".

But I am french so maybe it is a problem with my English ;-)

doom-fr, 2019-11-28:
When you will do that, I think it will be the occasion to add a similar functionality.
The possibility to define an area where it is not possible to draw trace, but possible to have plane.
It can be use to be sure to have a perfect full plane in a dedicated area.

doom-fr, 2020-01-17:
And also the opposite, define an area where there should be no plane (but where we can draw wire).
The aim is to not have plane too close to the pin of a TQFP48 (for example) on the inside side, but to allow connection of wire to that pin from the inside.
This is for managing things under the device.
(Do not know if it is really clear).

closed 2023-06-13 by ubruhin