LibrePCB/LibrePCB #599: Unittest: BoardPlaneFragmentsBuilderTest failed

ktrace, 2019-11-27:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.2
Git Revision: acdd94d9d2
Qt Version: 5.12.3
Operating System: gentoo
...
```

##### SUMMARY

BoardPlaneFragmentsBuilderTest fail

##### STEPS TO REPRODUCE

Build, run unittests.

##### EXPECTED RESULTS

All tests run OK.

##### ACTUAL RESULTS

BoardPlaneFragmentsBuilderTest failed.

##### ADDITIONAL INFORMATION

Console:
[ RUN      ] BoardPlaneFragmentsBuilderTest.testFragments
unknown file: Failure
C++ exception with description "The directory "/var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project" does not contain a LibrePCB project." thrown in the test body.
[  FAILED  ] BoardPlaneFragmentsBuilderTest.testFragments (1 ms)
[----------] 1 test from BoardPlaneFragmentsBuilderTest (1 ms total)

Trace:

> 
> write(1, "[----------] 1 test from BoardPl"..., 56[----------] 1 test from BoardPlaneFragmentsBuilderTest
> ) = 56
> write(1, "[ RUN      ] BoardPlaneFragments"..., 58[ RUN      ] BoardPlaneFragmentsBuilderTest.testFragments
> ) = 58
> statx(AT_FDCWD, "/var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project/.backup/backup.lp", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffcb2e24480) = -1 ENOENT (No such file or directory)
> statx(AT_FDCWD, "/var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project/.autosave/autosave.lp", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffcb2e24480) = -1 ENOENT (No such file or directory)
> statx(AT_FDCWD, "/var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project/.librepcb-project", AT_STATX_SYNC_AS_STAT, STATX_ALL, 0x7ffcb2e242f0) = -1 ENOENT (No such file or directory)
> write(1, "unknown file: Failure\nC++ except"..., 292unknown file: Failure
> C++ exception with description "The directory "/var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project" does not contain a LibrePCB project." thrown in the test body.
> ) = 292
> write(1, "[  FAILED  ] BoardPlaneFragments"..., 65[  FAILED  ] BoardPlaneFragmentsBuilderTest.testFragments (3 ms)

ubruhin, 2019-11-27:
Did you clone the Git repo recursively, i.e. including the submodule `tests/data/`?

And since the error message is pretty clear, did you already check if the directory "/var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project" exists and contains a LibrePCB project?

ktrace, 2019-11-28:
> $ ls /var/tmp/portage/sci-electronics/librepcb-0.1.2/work/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project
> boards  circuit  library  project  README.md  resources  schematics  test_project.lpp
all OK. tarball downloaded from https://download.librepcb.org/releases/...

dbrgn, 2019-11-28:
I can reproduce this:

```
wget https://download.librepcb.org/releases/0.1.2/librepcb-0.1.2-source.zip
unzip librepcb-0.1.2-source.zip
cd librepcb-0.1.2
mkdir build && cd build
make sub-tests-all -j 8
output/librepcb-unittests
```

Results in:

```
[----------] Global test environment tear-down
[==========] 1062 tests from 42 test suites ran. (13802 ms total)
[  PASSED  ] 1058 tests.
[  FAILED  ] 4 tests, listed below:
[  FAILED  ] DirectoryLockTest.testStaleLock
[  FAILED  ] SystemInfoTest.testIsProcessRunning
[  FAILED  ] SystemInfoTest.testGetProcessNameByPid
[  FAILED  ] BoardPlaneFragmentsBuilderTest.testFragments
```

Failures:

```
[ RUN      ] BoardPlaneFragmentsBuilderTest.testFragments
unknown file: Failure
C++ exception with description "The directory "/tmp/librepcb-0.1.2/tests/data/unittests/librepcbproject/BoardPlaneFragmentsBuilderTest/test_project" does not contain a LibrePCB project." thrown in the test body.
[  FAILED  ] BoardPlaneFragmentsBuilderTest.testFragments (1 ms)
```

```
[ RUN      ] DirectoryLockTest.testStaleLock
../../../tests/unittests/common/fileio/directorylocktest.cpp:268: Failure
Value of: success
  Actual: false
Expected: true
execvp: No such file or directory
[  FAILED  ] DirectoryLockTest.testStaleLock (1 ms)
```

```
[ RUN      ] SystemInfoTest.testIsProcessRunning
../../../tests/unittests/common/systeminfotest.cpp:96: Failure
Value of: success
  Actual: false
Expected: true
execvp: No such file or directory
[  FAILED  ] SystemInfoTest.testIsProcessRunning (1 ms)
```

```
[ RUN      ] SystemInfoTest.testGetProcessNameByPid
../../../tests/unittests/common/systeminfotest.cpp:125: Failure
Value of: success
  Actual: false
Expected: true
execvp: No such file or directory
[  FAILED  ] SystemInfoTest.testGetProcessNameByPid (1 ms)
```

I don't have the time right now to debug *why* they fail.

ktrace, 2019-11-28:
as I saw in strace test even not try to open/stat _test_project/test_project.lpp_

ubruhin, 2019-12-02:
It's a deployment issue. In the source archive all files/folders starting with a dot are removed (.git, .gitignore, ...) since those make no sense in the archive. Thus the `.librepcb-project` files in the test data are removed too, which breaks some tests.

But actually I wonder if we should even remove the whole `tests` directory in source archives? This directory will become larger and larger because of the test data (it is already 9.1MB, which is 26% of the whole source archive)...

dbrgn, 2019-12-02:
Yep, tests are probably not needed in the release archives.

ktrace, 2019-12-02:
In release sources? Needed of cause. Source-based distro make binary from it and test may be a part of build process. For example in gentoo: src_test() is a phase of build.

dbrgn, 2019-12-03:
Yes, but on the other hand the archives are not created by CI at all if the tests fail.

The main reason to run the tests again on the target system would be to ensure compatibility with shared libraries on the host system. What do you think about that, @ubruhin?

ktrace, 2019-12-03:
In sources-based distros app build on clients PC, so I think tests will be good idea. So I'm planned build with system quazip, linked dynamically and better test after :) And I think many people who download sources prefer tests. Other hand tests may be in separate archive, but I think it's not very simple solution...
And I had another idea: separate target "test" in build system with build and run tests. Same way used in Gammu project and I saved a lot of time and nerves  because of it. But I think need new feature request for it (and for system quazip too :))

ubruhin, 2019-12-05:
Yeah basically it's a good idea to run tests to check compatibility with the system libraries etc ;)

What about keeping (and fixing) the tests in source releases as long as their file size doesn't really hurt, and think again about this once tests require >>30MB or so?

> And I had another idea: separate target "test" in build system with build and run tests.

Sounds reasonable from the users/packagers point of view. But as a developer actually I really like that tests are always built too, since this way you immediately get compile errors if you break something (avoid wasting time for waiting until CI complains about compile errors). Maybe we could make it configurable somehow to decide whether tests should be built or not...

dbrgn, 2019-12-06:
> What about keeping (and fixing) the tests in source releases as long as their file size doesn't really hurt, and think again about this once tests require >>30MB or so?

:+1:

ktrace, 2019-12-06:
> What about keeping (and fixing) the tests in source releases as long as their file size doesn't really hurt, and think again about this once tests require >>30MB or so?

It's not problem for sources tarball. It'a life, it's a reality: sources always grow.

> But as a developer actually I really like that tests are always built too

Hmm. make && make tests - two actions. make (full with tests) && run tests binary - two actions. See no difference :) Am I wrong?

ubruhin, 2019-12-06:
> Am I wrong?

I'd like to also provide the behavior of "make && make tests" within QtCreator since this is the "official" IDE for Qt projects :wink:

ubruhin, 2019-12-22:
I adjusted my deployment script, so this issue should be fixed in the next release.

closed 2019-12-22 by ubruhin