LibrePCB/LibrePCB #600: Board editor : draw a trace segment by segment, not 2 segments at each click

doom-fr, 2019-11-27:
##### SUMMARY
When you draw a connection, the "draw trace" tool show the already drawn segments, and the two next ones. When you click, it draws definitely this 2 segments and show 2 new segments.
It is therefore difficult to align with the target.

##### SOLUTION
Do exactly the same but only for 1 segment each time you click.
If you want to draw the 2 segments at the same time, just double-click.

dbrgn, 2019-11-27:
I think I don't understand this report. Maybe you can post a screenshot?

doom-fr, 2019-11-27:
difficult with a screenshot as it is dynamic.

When you draw a trace, show the two next segments, but only draw the first one when clicking.
It will be helpful to align the trace with other trace or device.

doom-fr, 2019-11-27:
[image]

doom-fr, 2019-11-29:
[image]

doom-fr, 2019-11-29:
[image]

dbrgn, 2019-11-29:
If you draw straight, you can create a single segment.

doom-fr, 2019-11-29:
I have other connections to do ...
And it not a real case. just an example to show.

doom-fr, 2019-11-29:
With the actual method, you have 2 things to align at the same time, and sometimes they are far away so you can zoom a lot.
With my proposition, you align the first thing then align the second one.
It is easier and faster.

If you don't want to change it for everyone, please set it as an option.
For me, it is a big HMI improvement.

doom-fr, 2020-05-30:
@ubruhin I want to point this issue because it is needed to improve the user experience.
All the more, it seems to be a little change in the code.

If you want to keep actual behavior also, maybe add a checkbox or everything else to switch this behavior **on** or **off**.

It will enhance user experience.

And if you can provide also to snap a pad if the cursor is on it, even if not on the grid, this will be the icing on the cake !

But alone, allow to draw wire segment by segment is already a great improvement.

From memory, this behavior is in Kicad, Altium and maybe many more other softwares.
There might be a reason for that !
And it is a good point to help users to switch to LibrePcb.

AvetosDesign, 2026-09-16:
KiCad also draws two segments at a time.  It does, however, recognize when one of those two segments is an extension of another one, and it merges them together.  I wonder if investing in a simple trace optimization algorithm (applied when a segment is committed) would accomplish the same goals.