LibrePCB/LibrePCB #603: Schematic editor : Prevent using invisible character in net names

doom-fr, 2019-11-27:
##### SUMMARY
I just see that my project has 2 different gnd.

The first one that is "gnd"
[image]

And the second one that is "gnd\n"
[image]

[image]

##### SOLUTION

Raise ERC if invisible characters are used in net names

doom-fr, 2019-11-27:
Get this error just after having change "gnd\n" to "gnd".
At the first try to put a via.

[image]

ubruhin, 2019-12-02:
Whooops, that should actually not be possible! It seems that somehow linebreaks are not rejected, but all other invalid characters are rejected properly :(

I'll fix this. But with the Bugfix, LibrePCB will reject opening projects which already contain linebreaks in net names. So you would need to manually rename the net to something else (a unique net name which is not used already).

doom-fr, 2019-12-02:
I have already removed the line break.

closed 2019-12-27 by ubruhin

rnestler, 2020-02-12:
> I'll fix this. But with the bugfix, LibrePCB will reject opening projects which already contain linebreaks in net names. So you would need to manually rename the net to something else (a unique net name which is not used already).

Would we need to provide migration paths for such things?

doom-fr, 2020-02-12:
I think that the probabilty to encounter this problem is very low.

So just documenting in this issue, where and how to change the errored net names will be enough.

D.

PR #637 (2019-12-27): Fix regex of various constrained strings
The regex to check following constrained strings allowed a trailing newline by accident (see #603):

- AttributeKey
- CircuitIdentifier
- ComponentPrefix
- ComponentSymbolVariantItemSuffix
- GraphicsLayerName
- Uuid

Fixed the regex by using "\A" instead of "^" and "\z" instead of "$".
Also added some unit tests to check for correct behavior.

Fixes #603.