LibrePCB/LibrePCB #604: Board Editor : Differentiate visible grid and snap grid

doom-fr, 2019-11-29:
##### SUMMARY
In board editor, the drawn grid is the same as the snap grid.
Problem is that snap grid needs to be little (⩽0.1mm)
So it is difficult to see where we are and the grid is very dense.

##### SOLUTION
Draw a visible grid higher than the snap grid.
It won't be a problem as it is easy to divide by 5 in our brain.
Here is a proposition : the big strokes are 10 times the snap grid and little ones 5 times.
Snap grid is represented by the 5 little squares (it is just an example to show the snap grid)

[image]

doom-fr, 2019-11-29:
And also, can you please set snap interval size as the first selected parameter when we open the grid window (as it is the mostly changed value) ?

plocher, 2020-01-25:
In Grid settings, think about supporting the following concepts ([__] is an input field for user preferences):
Snap Grid [__]  [units]
Major Marking [__]  dots or [__]  lines every [__] x Snap Grid units [__]  size [__]  color
Minor Marking [__]  dots or [__]  lines every [__] x Snap Grid lines [__]  size [__]  color

This allows for settings like
snap grid 0.100 mil
Major thick grey lines every 10x snap grid setting (i.e., every 1 inch)
Minor small white dots every 1x snap grid setting (0.100 inch)

ubruhin, 2025-12-18:
I have made a change to grid in #1647. I don't know if this will be the final solution, maybe there are still ideas to improve it. But I'd like to gather some experience now with the new system first.

closed 2025-12-18 by ubruhin