LibrePCB/LibrePCB #608: Board editor : Change color of inner layer 2 (to close of inner layer 1 color)

doom-fr, 2019-11-29:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.3-unstable
Git Revision:     bd616e14d1366486cf53cad8b8c7e3e67119d960
Build Date:       2019-11-17T15:33:39
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
inner layer 1 & 2 are quite same color

##### STEPS TO REPRODUCE
1. set to 2 inner layers
2. draw something in each layer

##### EXPECTED RESULTS
all copper layers (at least the 4 firsts) should have different colors that are easily distinguishable.

##### ACTUAL RESULTS
inner layer 1 & 2 are quite same color

[image]

##### ADDITIONAL INFORMATION
And when inner1 is selected, it is really quite the same as inner2

[image]

##### SOLUTION
exchange between inner layer 2 color and inner layer 4 color.
that will give this stack of color : 
[image]

doom-fr, 2019-11-30:
editing `boards/board/settings.user.lp` file allow to change all colors.

ubruhin, 2020-02-16:
@dbrgn Since you are the master of colors, maybe that's an issue for you? :grin:

closed 2023-10-18 by ubruhin