LibrePCB/LibrePCB #612: Board Editor: Segment vs Track

EdizonTN, 2019-12-01:
##### VERSION / OS / ENVIRONMENT
LibrePCB Version: 0.1.3
Git Revision:     56bc60d347ff67df0fe1d57807d03f0606de557f
Build Date:       2019-11-30T21:59:40
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Windows 7 SP 1 (6.1)
Platform Plugin:  windows

##### SUMMARY
In PCB Editor is swapped thinking about track and segment of track.

Wrong (actual situation): 
Trace = part of segment.
Segment = finished connection between pads.

Correctly (expected):
Trace = finished connection between pads.
Segment = single part of trace.

##### STEPS TO REPRODUCE
Open PCB editor of routed PCB, and right click to part of trace.
Click "remove trace" - LibrePCB clear a part of the trace (segment).
Or
Click "remove whole segment" - LibrePCB clear whole trace not segment.

##### EXPECTED RESULTS
Trace: pad to pad connected copper (one or more segments = complette trace)
Segment: simple part of trace.
SEGMENT:
[image]

TRACE:
[image]

##### ACTUAL RESULTS
[image]

ubruhin, 2019-12-01:
Hi @EdizonTN,

Sorry for not responding yet on Transifex. It's still on my ToDo list to review your comments and fix the source strings where needed ;)

Regarding the naming: I agree it might be a bit confusing. The thing is, we call it "**net** segment", not "**trace** segment" or something like that. So it is a part of the net, not a part of a trace.

Basically there are three things to differentiate, for example when selecting elements:
- The whole net, i.e. all elements of the same net (currently called just "net")
- Several connected lines & vias, building a part of a net (currently called "net segment")
- Only a single line (currently called "trace")

This applies to both, the schematic and the board editor, except that single lines are called "net line" instead of "trace".

But yeah, probably we could somehow improve the terminology, although I'm not sure what the best naming would be...

Btw, thanks a lot for doing the translations @EdizonTN, you did a fantastic job on Transifex! :tada:

doom-fr, 2019-12-02:
I agree with definitions of @EdizonTN .

A segment is the part between nodes.
A trace is a whole connection between pads (stopping at the planes).
And the net can be every segments with this net.

If you say "Add Trace" for adding multiple lines, the group of lines is a trace.

And in Mathematics, a segment (line segment) is "a part of a line that is bounded by two distinct end points, and contains every point on the line between its endpoints" (Wikipedia).

dbrgn, 2019-12-03:
> A segment is the part between nodes.
> A trace is a whole connection between pads (stopping at the planes).
> And the net can be every segments with this net.

I think I also agree with those terms.

ubruhin, 2020-02-26:
Fixed in #674.

closed 2020-02-26 by ubruhin