LibrePCB/LibrePCB #614: Board Editor: Cannot change routing layer

EdizonTN, 2019-12-01:
##### VERSION / OS / ENVIRONMENT
LibrePCB Version: 0.1.3
Git Revision:     56bc60d347ff67df0fe1d57807d03f0606de557f
Build Date:       2019-11-30T21:59:40
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Windows 7 SP 1 (6.1)
Platform Plugin:  windows

##### SUMMARY
Cannot change a routing layer during route a track, when route started from a SMD pad.

##### STEPS TO REPRODUCE
[image]

##### EXPECTED RESULTS
I expected change the layer :-)

##### ACTUAL RESULTS
Layer won't changed

doom-fr, 2019-12-02:
There's a little bug on this, but it is a blocking bug.

First, I am pretty sure your vias are not attached to the traces.
If you try to move the via, if it is attached, the trace moves also.
Via are not auto connected to traces, so you have to place the vias and then connect them.

After, this is done, if you try to make a trace on another layer, it works.

EdizonTN, 2019-12-02:
> First, I am pretty sure your vias are not attached to the traces.
- Why? I don't change grid between drawing traces and placing a vias.
If trace and via are on-grid, both must be connected. If no, grid is without sense (or snaping to grid).

> If you try to move the via, if it is attached, the trace moves also.
- OK, If via is not connected to trace, but is placed over this trace = error. (will be good visualise this error immediatelly)

> Via are not auto connected to traces, so you have to place the vias and then connect them.
- Sorry but this is a wrong. If I draw trace and change layer during, vias have to autoconnect.

> After, this is done, if you try to make a trace on another layer, it works.
- I'll try evening..

EdizonTN, 2019-12-03:
Yes @doom-fr, looks it as a problem with not correctly connected vias. But I have no idea how it connecting correctly durng routing....

I use same grid for route and place via. Why is via after place to route not connected?

doom-fr, 2019-12-03:
I think just because it is not still done (maybe in 0.1.4 ;-) ).
The only workaround I found is to place the via before connecting it.
It is perfectible but it works.

closed 2019-12-03 by EdizonTN

doom-fr, 2019-12-03:
You're welcome !

doom-fr, 2019-12-03:
And you can also give your feedback here : https://librepcb.discourse.group/t/board-editor-improving-via-insertion/161