LibrePCB/LibrePCB #615: Use Weblate for translations

comradekingu, 2019-12-03:
Hello, I want to help translate and improve the strings in LibrePCB, but can't use Transifex since it takes ownership in my work doing so. It is also a bad closed source tool.

https://weblate.org/ is my preference, and the equivalent Hosted Weblate variant is gratis to use for libre software projects, as opposed to Transifex, (which doesn't allow any venue of income).

dbrgn, 2019-12-03:
> but can't use Transifex since it takes ownership in my work doing so

I'm not aware of any rights transfership in their terms of service. Can you point it out?

comradekingu, 2020-03-31:
@dbrgn 

> You grant Transifex a non-exclusive, world-wide, perpetual, royalty-free, irrevocable, sublicensable and transferable license to use your words for purposes of performing the Services. You agree to indemnify Transifex for any third party claims, demands or losses arising as a result of your failure to obtain all such necessary rights, permissions and authorizations.

> The Services provide features that allow you to share your words with others or to make your words public.

dbrgn, 2020-03-31:
Hm, I don't understand. First of all, if you contribute to an open source project (whether on Transifex or not), you already do grant *everybody* a non-exclusive, world-wide, perpetual, irrevocable license for practically anything. The "royalty-free", "sublicensable" and "transferable" parts depend on the actual license used.

*Of course* you need to grant Transifex the permission to share your words with others (e.g. the open source project maintainers, or other contributors).

The whole *idea* of Transifex is "to share your words with others or to make your words public" (at least when using it with an open source project).

comradekingu, 2020-03-31:
You had me at you don't understand. You are confusing so many things it seems as if you are going by your initial idea. Will get around to detailing my view later on.

dbrgn, 2020-03-31:
@comradekingu Please stay friendly. You are of course welcome to detail your views. However, please refrain from using wording like "You had me at you don't understand".

I'm not generally opposed to using weblate (using open services is almost always better to avoid lock-in), but someone would have to do the work of setting up a project, updating the contributors documentation, updating the developer documentation, updating the continuous integration build servers (which pull from transifex using the `tx` CLI application) and then testing all that. There are also features we use in Transifex like glossaries (to ensure consistent translations of key terms) that would need to be ported to Weblate. Furthermore, existing translators would need to be notified that we are moving to a different platform.

@comradekingu would you offer to do all this work?

comradekingu, 2020-03-31:
I am sorry. No harm intended.
Yes.

dbrgn, 2020-03-31:
Before we take any action here, I'd still be interested in your view on what the Transifex EULA means in practical terms for LibrePCB. Afterwards, @ubruhin would need to decide whether he'd be willing to do the effort, since a lot of the work (testing release builds) will be stuck with him.

comradekingu, 2020-03-31:
Having to be 18 years old strikes me as an otherwise arbitrary restriction, to start somewhere.
Avoiding all the tracking scripts TX has is another fine reason to abandon ship too.
Weblate has a lot of checks, and grants a much better overview, and has more libre software translators. I have only seen increased activity and fixed bugs for projects making the move.

dbrgn, 2020-03-31:
> Having to be 18 years old to contribute strikes me as an otherwise arbitrary restriction, to start somewhere.
> Avoiding all the tracking scripts TX has is another fine reason to abandon ship too.

Those are all good reasons! However, you wrote explicitly that:

> it takes ownership in my work

But you did not substantiate that claim so far. You then quoted their license grant, which grants them a license, *not* ownership. From Transifex terms of service:

> You retain full ownership of your words, and Transifex has no ownership rights in your words. These Terms do not grant us any rights to your words or intellectual property except for the limited rights that are needed for Transifex to run the Services, as explained below.

Hosted Weblate also has [terms of service](https://weblate.org/en/terms/) that you need to accept. They *also* include a license grant:

> Should the Project opt in for the Translation Memory, license to use the translation is granted to all Translation Memory users.

The intent for Transifex and Weblate is probably the same. Since it's a hosted service and you upload copyrighted material to their website, you *need* to grant them the license to do something with that content (since they need share this content with other users of the platform, e.g. if you invite translators that don't themselves have any ownership of these strings).

The strings license you grant to the LibrePCB project when translating (currently GPL) is actually more broad, since LibrePCB or *any other party* (including Transifex, even if we don't use them) may use the strings for any purpose as long as the GPL is followed, while Transifex (under their terms of service) may only use them to provide their translation services.

(Btw, GPL is not a good license for strings. We should think about using a license like CC-BY or CC0 for strings.)

To summarize: If you still think that Transifex takes ownership of your work, please substantiate that claim. However, even besides that, there may be good reasons to switch to Weblate or another service (like tracking, which I dislike). I am *not* saying Transifex is better than Weblate in any way (I was simply not familiar with Weblate so far). However, then it's a question of tradeoffs and priorities, and not of ownership transferal.

comradekingu, 2020-03-31:
I never said ownership is taken away, I said they "take ownership in". I am technically right on any other level than one could contest on account of a technicality. Albeit so;
Yes, it is a license, because I can't actually legally waive my copyright this way. It is semantic to the tune of "a license that emulates ownership" being the exact same result, more accurately described.

Weblate specifically lets you use the TM, _without_ contributing to it. It only has:

> Should the Project opt in for the Translation Memory, license to use the translation is granted to all Translation Memory users.

Further it (contrary to TX) doesn't have special terms for Europeans, it is the same for everyone.
You can set a translation license differing from the project license. It also has no caveats for sharing with third parties. It is only hosted on servers in Europe, when TX

(Crowdin actually requires participation in the TM to use it as a "free" service for Open Source".
One of the biggest problems with both is that they don't allow a project to have any means of revenue to apply for it.

TX does step in to monetize "free" accounts when they wish, so steer clear of that. It is _not_ cheap, and their commercially offered service is famously bad.

I don't agree that the TX terms amount to "Should the Project opt in" amounts to what TX has.
"The intent for Transifex and Weblate is probably the same." I find wanting, and unsubstantiated beyond what needs to be common on a base level. TX is a different beast.

TX caveats:
>We may share information from time to time to our offices or personnel, or to third parties, located throughout the world, and may be viewed and hosted anywhere in the world, including countries that may not have laws of general applicability regulating the use and transfer of such data. To the fullest extent allowed by applicable law, by using any of the above, you voluntarily consent to the trans-border transfer and hosting of such information.

American law lets the NSA snoop as it pleases there.

On Weblate you can delete any and all of your strings, and they actually disappear. Not so on TX.
TX is not serves them as suggestions for other strings even if they are deleted, but also doesn't care about GDPR claims, saying translations only happen between actors, and has no system of exporting it. Couldn't be more different.

TX however makes this false claim:
>You have the right to access the information we hold about you. You can contact us and we shall inform you about:

The GPL is a restrictive copyleft license, so I am not buying that it is broader. In the very least not at all in terms of things that I don't want, which is what TX does with its increased scope.

> 4. How do we share personal information?
> 
> We may share personal information as follows:
> 
>     We may share information to technical service providers which may operate the technical infrastructure that we need to provide the Transifex Service, in particular providers which host, store, manage, and maintain the Transifex application, its content and the data we process, including translations. We also use technical service providers to help us communicate with you. As of the date this policy went into effect, we use RackSpace storage service to store some of your information (for example, your Words). We use Chargebee and Stripe to manage your online payment service.

Various sections of the "Privacy Policy":
>We may share your personal information with third-party services integrated in our platform. Using such services may require that you create a separate account with such third parties. Upon doing so their own Terms of Service and Privacy Policy shall apply.

>We may share your personal information with other companies of the Transifex Group to carry out our daily business operations and to enable us to maintain and provide the Transifex Service to you.

>We also may share personal data with certain marketing and advertising partners to send you promotional communications about Transifex.

>We may share aggregated or anonymized data. We may disclose or use aggregated or anonymized data for any purpose. An example would be for marketing, analytics or research purposes.

>We may share other information with consent with third parties when we have consent to do so.

>We may share information per our customers’ instructions.

>We may share information to protect the rights and property of Transifex, our agents, customers, and others. This includes enforcing our agreements, policies, and terms of use.

> [2. How We Use Personal Information](https://www.transifex.com/legal/privacy/)
 To link or combine user information with other personal information.

> [Your profile includes, but is not limited to](https://www.transifex.com/legal/terms/),

> [The Services may continue to change over time as we refine and add more features.](https://www.transifex.com/legal/terms/)

On TX you cannot be anonymous 
> You represent and agree that all information you provide to Transifex in connection with your access to and use of the Services is, and shall be, true, accurate and complete to the best of your knowledge, ability and belief.

(from Terms)
> You will update and correct information you have submitted to the Services and ensure that it is accurate at all times (out-of-date information will invalidate your account).

It wants to advertise based on spying
[... cut ...]
>please contact us at privacy@transifex.com or at 94 Solaris Avenue, Camana Bay, P.O. Box 1384, Cayman Islands KY1-1108.

Cayman Islands… it is otherwise a Limited company with an US address.

I didn't list all the egregious points in the "terms" or "cookies policy", but it isn't better there either.
You need to read it all to know how to even opt out, manually, etc etc.
It is _a lot_ of legalese spaghetti text. In terms of cookies: TX uses Google Analytics and a whole slew of others, Weblate uses self-hosted matomo.

ubruhin, 2020-03-31:
Hi @comradekingu,

I totally agree that from the privacy and transparency point of view, Weblate would be much better than Transifex. If we would set up a new project today, I'd definitively give Weblate a try first. But unfortunately, things are more complicated because we already set up our translation workflow with Transifex 2 years ago. As @dbrgn mentioned, high effort would be needed to migrate our workflow to a different translation service. And like for every issue, in the end it's always a question of prioritizing... ;)

Beside that, I see some open questions:
- From the technical point of view, does Weblate satisfy our needs? Are there any features missing? Or are there even useful features which we don't have on Transifex?
- Does it integrate well with our current workflow where the "source of truth" of our translations are stored in the Git repository [librepcb-i18n](https://github.com/LibrePCB/librepcb-i18n)? Including support for release branches? Note that we maintain separate translation files for each release branch, and there was quite some configuration effort needed to set this up.
- Could we easily test Weblate, or do we already have to send a manual request for the free hosting service?
- On the website, it's not clear what we get for free:
  > Source strings: Negotiable
  > Languages: Negotiable
  > Projects: 1
  
  Do we get everything we need? Is 1 project sufficient? Do we get some kind of commitment for the time span how long we get this free service (e.g. I don't like to submit a new request every year or so)?
- And for me the most important thing: Would our current translators agree with moving to Weblate? I don't like to annoy our translators. Probably most of them started using Transifex especially for translating LibrePCB. Maybe some of them would be annoyed when migrating to Weblate. IMHO the translators are much more important than the tool used - if they don't agree with using a different tool, I don't agree too.

comradekingu, 2020-04-02:
@ubruhin Transifex doesn't handle fuzzy strings, which is infuriating.
I don't know exactly how the versioning is set up, but I have never seen that work on TX. I think maybe it does 100% matches only, if memory serves me right.

Weblate has links to source strings, and context on offer. For all accounts.

A lot of projects have separate string repos, an alternative to that is squashing commits. Per author, or language, etc.

Service is gratis for libre software projects. I wouldn't reccomend it if it was some kind of trap (like TX).
I'll pay it if it somehow ends up costing something. This offer will not all of a sudden expire.

The best libre software translators are on Weblate already. If you value translators give them a better tool to work with.

ubruhin, 2025-05-25:
It has been a long time until this was discussed. I think many things have changed in the mean time and it might be worth to take a look at this again. For example, our release process is now a bit simpler and I think we can ignore release branches for now regarding translation toolchain.

Also I really agree we should move away from proprietary Transifex - as long as Weblate (or another free translation tool) fulfills our requirements.

To not bother our current translators too much, I have an idea of a possible migration scenario:
- First of all, evaluate which tool would be the best and clarify all the question already mentioned above in this thread.
- Then set up the new translation workflow **in parallel** to the current Transifex worflow. To not cause any conflicts and to not bother our current translators, only configure it for new languages for now, not for languages we already have on Transifex. So all the current languages would be translated with Transifex, and all new languages with a new tool.
- Gain some experience with the new tool set up this way, to see if it really works as intended.
- Once we are happy with the new tool, slowly start to migrate more languages from Transifex to the new tool. This shall be done only with the agreement of our current translators. If there are languages where the majority of translators don't support the migration, we can keep them on Transifex for more time.
- Once we really have all languages migrated, close our Transifex project.

For the tool evaluation and requirement analysis, I'd be happy for help so I added the "help wanted" label. Also for example for Weblate the question is if we can use their hosted service for free or if a self-hosted Weblate would be the best option (though I really want to keep server administration efforts low). Or are there even other hosters for Weblate which offer the service for free? I have in mind Codeberg also offers something like that, but it might not be Weblate.