LibrePCB/LibrePCB #617: ControlPanel : Opening many projects instead of just the one selected

doom-fr, 2019-12-03:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.4-unstable
Git Revision:     ac1eff31f2f4b71e6ec9997cae7280c6d1d744e2
Build Date:       2019-12-01T23:04:54
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
When I open a project, it opens sometimes 2 projects, the one requested and another one.

##### STEPS TO REPRODUCE
1. having multiple projects
2. open LibrePCB
3. open a project

##### EXPECTED RESULTS
The requested project is open

##### ACTUAL RESULTS
The requested project is open and sometimes also another one.

##### ADDITIONAL INFORMATION
no

dbrgn, 2019-12-03:
I suspect you're double-clicking on a project entry that is not the latest.

[image]

When you click on the second project (demo.lpp), it will move up in the list. If you click again, then the desklift.lpp project will have moved down, and will open as well.

TL;DR: Only single-click, don't double-click.

@ubruhin maybe we could avoid this somehow?

doom-fr, 2019-12-03:
OK, I think your right.

The problem is that during the delay to open the project, the user don't know something is going on.
Maybe add a sandglass can help in this.

We can also consider to open a project with a double-click.

ubruhin, 2019-12-03:
Duplicate of #293

closed 2019-12-03 by ubruhin