LibrePCB/LibrePCB #618: Board editor : Plated Through Holes

doom-fr, 2019-12-03:
##### SUMMARY
Currently, it is not possible to place Plated Through Holes.
It can be done by adding a via, but it is so connected to a net.

##### SOLUTIONS
I propose three solutions :
1. add a PTH tool
2. add a 'no net' choice to via
3. in "add hole" tool, add a 'Plated' checkbox

I think, solution 3 is the best one.

Maybe also thinking about merging VIA, NPTH and PTH in a single item (and add a constraint : if connected to a net, it should be plated).

doom-fr, 2019-12-03:
I found a dirty workaround : 

1. create an unused net
[image]

2. and add a via with this net
[image]

The problem is that it is dirty and you have to create an unused net for each PTH.

dbrgn, 2019-12-03:
Do you want silkscreen over your via or not?

If not, you could create a device in the library with a single through-hole pad.

doom-fr, 2019-12-03:
I do not take it in account.
So it is better for me to do like you say.
But creating this component just for that is also dirty (maybe more), so it remains a workaround.
Thanks !

doom-fr, 2019-12-04:
In fact, it seems to be a little bit more complicated that vias are covered by solder mask.
(@dbrgn I think you want to say 'solder mask' instead of 'silkscreen' in your last message).

Via objects are covered by solder mask if it is inside a plane with the same net, so if it is a real via.

In my workaround, it is not the case, so it is not covered.

dbrgn, 2019-12-04:
Yes, I meant soldermask :slightly_smiling_face:

ubruhin, 2021-02-12:
Yeah I think it makes sense to support placing vias with no net :+1: 

I'd not extend holes with a "plated" attribute since PCB manufacturers usually are not able to create plated holes with no restring at all.

Currently, vias are represented this way in a board file:

```scheme
 (netsegment 02c00bef-5bd6-454d-801c-6ec49ff34d92
  (net bc37feba-6bec-419e-ac2c-81beb725d5dc)
  (via 1cadb9f9-6c50-4590-ad7e-87d9238e2819
   (position 24.009 9.699) (size 0.7) (drill 0.3) (shape round)
  )
 )
```

Currently I think we should implement unconnected vias exactly the same way - nested within a netsegment - but simply allow the net to be none:

```scheme
 (netsegment 02c00bef-5bd6-454d-801c-6ec49ff34d92
  (net none)
  (via 1cadb9f9-6c50-4590-ad7e-87d9238e2819
   (position 24.009 9.699) (size 0.7) (drill 0.3) (shape round)
  )
 )
```

Basically this would even allow to create traces with no net, which might be nice as well (although I never needed such a feature...). I think I'll quickly try to implement this to see if this causes some serious issues.

PR #854 (2021-02-18): Allow creating unconnected board net segments (traces/vias)
This allows to create traces and vias which are not assigned to any net. The main purpose is to provide the ability to create plated through holes for mechanical purposes only. A positive side effect is that the "add via" tool in the board editor is no longer rejected if the schematic is empty (i.e. no nets available) - now this tool works as usual since it is able to create unconnected vias.

[image]

Fixes #618.
closed 2021-02-18 by ubruhin