LibrePCB/LibrePCB #620: Fabrication data : Add a Aisler compatible predefined config

doom-fr, 2019-12-04:
##### SUMMARY
There are actually 2 predefined config : 
1. Default
2. Protel naming

None of these is compatible with Aisler which is a partner of LibrePCB.

##### SOLUTION
Add a 3rd predefined config for Aisler with good naming.

dbrgn, 2019-12-04:
For Aisler, you should actually not send them Gerber files. Instead, upload the `*.lppz` archive directly. You'll be able to verify your design in their preview, which is guaranteed to match the manufactured PCBs.

Note: Currently they still run an old version of LibrePCB which is not yet compatible with the `*.lppz` files, but that should be updated in a few days (I hope). In doubt, ping @PatrickFranken or send them an e-mail :slightly_smiling_face: 

In summary, since the Gerber export is not meant to be used with Aisler, I don't think it makes sense to add an Aisler config to the exporter.

(AFAIK @ubruhin also planned to add support for "manufacturer-configs", which include things like export format, design rules, etc. That's not yet implemented though.)

closed 2019-12-04 by dbrgn

PatrickFranken, 2019-12-04:
@doom-fr we are just in the process of updating LibrePCB to 1.3 on our servers. Should be done tomorrow so you can just drag and drop the your LibrePCB project.

doom-fr, 2019-12-05:
@PatrickFranken , can you please reply to my emails ?
I am in a hurry. Thanks.

PatrickFranken, 2019-12-05:
Just got an answer ;)

PatrickFranken, 2019-12-07:
Alright, we finally updated LibrePCB on our servers. Happy fabbing!