LibrePCB/LibrePCB #621: Fabrication Data : generate device placement file (position and rotation)

doom-fr, 2019-12-04:
##### SUMMARY
Pick and Place machines need a file with position and angle of all devices on the board.
This file is currently not generated by the fabrication data generation.

##### SOLUTION
Generate the file during fabrication data generation.
If there is components on bottom, another file should be generated.

##### ADDITIONAL DATA

The file look like that : (example generated by Kicad)
```
# Ref     Val       Package     PosX        PosY     Rot         Side
C11       22uF      0805        51.2000    -22.60      0.0000    top 
C12       22uF      0805        56.1000    -32.00    270.0000    top 
C13       100nF     0805        52.0000    -25.70    270.0000    top 
C14       100pF     0805        51.0000    -28.80      0.0000    top 
C21       47nF      0805        33.9600     -7.87    180.0000    top 
C22       220nF     0805        36.2500    -11.37    270.0000    top 
C23       22uF      0805        36.1000    -25.40    270.0000    top 
C24       2.2uF     0805        33.8000    -21.00    180.0000    top 
```

EdizonTN, 2019-12-05:
I absolutly agree with idea to generate placement data, but i have some notices.

Placement of the components (ex. reflow soldering) is not a part for fabrication process. It's a part of assembly process, and can be done in another company as fabrication. And into assembly house you will not send gerber data of course... :-)
Due this, is better, creating Pick&Place data package with BOM and Assembly prints, as independend process.

My proposal is create new Menu->File item "Generate Assembly data" which open a new window with options to export:
- BOM 
- PDF printout of Top Assembly
- PDF printout of Bot Assembly
- PDF printout of additional notification text (specialities in assembly process). All printouts can be in one PDF file, each on new side.
- Generate Pick&Place data (csv file)

Output path as "./output/{{VERSION}}/Assembly/{{PROJECT}}"

And Also Fabrication path will be good to little bit modified as "./output/{{VERSION}}/Fabrication/{{PROJECT}}"

And please thinking also, about option to saving THT and SMT devices to separate files.

Attachment - Another example of Pick&place .csv file content:
"Designator","Comment","Layer","Footprint","Center-X(mm)","Center-Y(mm)","Rotation","Description"
"IC12","PCA9955B","BottomLayer","HTSSOP65P640X970-28N_-_SOT1172-3_-_MO153","2.857","8.255","90","16-Channel LED CC driver with PWM current controller"
"IC11","PCA9955B","BottomLayer","HTSSOP65P640X970-28N_-_SOT1172-3_-_MO153","-8.572","8.255","90","16-Channel LED CC driver with PWM current controller"
"TVS1","-7 to 12V","TopLayer","SOT95P230X110-3N_-_SOT-23_-_TO-236AB","-12.065","-4.128","360","TVS dual 600W, for RS485"
"_XX5","FIDU-Mark","BottomLayer","FIDU_Square_Mark_1mm","15.557","-14.922","90","Placement Square mark for assembly machine"
"_XX6","FIDU-Mark","TopLayer","FIDU_Square_Mark_1mm","15.557","-14.922","90","Placement Square mark for assembly machine"

ubruhin, 2019-12-22:
I searched a bit in the Internet, but found almost no usable information about pick&place file formats. It's a mess, there seems to be no standard file format or specifications available :(

Does someone know if the positions for the bottom devices should be mirrored, or if we should always use positions from the top view (i.e. looking "through the board")?

I started to implement a pick&place export, the current output looks like that:

```csv
# Pick&Place Position Data File
# 
# Project Name:        Hydro Battery Charger
# Project Version:     v2
# Board Name:          default
# Generation Software: LibrePCB 0.1.4-unstable
# Generation Date:     2019-12-22T21:37:19
# Unit:                mm
# Rotation:            Degrees CCW
# Board Side:          Top

Designator,Value,Device,Package,Position X,Position Y,Rotation,Side
C1,LGW2E471MELZ40 470μF/250V,Elko 22/10mm,Elko 22/10mm,40.005,48.26,180.0,Top
C2,LGW2E471MELZ40 470μF/250V,Elko 22/10mm,Elko 22/10mm,62.5475,48.26,180.0,Top
C3,LGW2E471MELZ40 470μF/250V,Elko 22/10mm,Elko 22/10mm,40.005,11.43,0.0,Top
C4,LGW2E471MELZ40 470μF/250V,Elko 22/10mm,Elko 22/10mm,62.5475,11.43,0.0,Top
C5,10nF,C-0603,C-0603,117.475,26.035,270.0,Top
C6,2.2μF 25V,C-0805,C-0805,63.8175,27.735074,180.0,Top
C7,220nF,C-0603,C-0603,123.19,22.86,0.0,Top
C8,220nF,C-0603,C-0603,119.6975,25.0825,180.0,Top
```

Any thoughts?

doom-fr, 2019-12-22:
I think the easier would be to have a try on different other EDA software to look at what they do.
Unfortunately, Kicad is silly on my PC. So I can't try by myself.

doom-fr, 2019-12-22:
I found this [document](http://i.screamingcircuits.com/docs/Understanding%20the%20Centroid%20file.pdf) : 

> Location:
> The “LocationX” and “LocationY” values describe the part’s offset from the board origin. The
> location values require that the part origin be centered in the part. The board XY origin of 0,0 is
> in the lower left corner of the board. The 0,0 origin for the bottom of the board is in the lower left
> corner, looking at the top of the board, though the board. Preferred units are in inches (0.0000”).

EdizonTN, 2019-12-22:
Hi,

I create demo PCB in Commertcial EDA Tools, and my discovery is: Positions are NOT mirrored.

**Example board:**
[image]

**CSV Output:**
```
Pick and Place Locations
C:\PCB_Test\Project Outputs for PCB_Test\Pick Place Output\Pick Place for main.csv

========================================================================================================================
File Design Information:

Date:       22/12/19
Time:       22:19
Revision:   Not in VersionControl
Variant:    No variations
Units used: mm

"Designator","Comment","Layer","Footprint","Center-X(mm)","Center-Y(mm)","Rotation","Description","ComponentKind"
"_XX6","FIDU-Mark","BottomLayer","FIDU_Square_Mark_1mm","10.0000","10.0000","90","Placement Square mark for assembly machine","Standard"
"_XX5","FIDU-Mark","BottomLayer","FIDU_Round_Mark_1mm","-10.0000","10.0000","90","Placement Round mark for assembly machine","Standard"
"_XX4","FIDU-Mark","BottomLayer","FIDU_Round_Mark_1mm","10.0000","-10.0000","90","Placement Round mark for assembly machine","Standard"
"_XX2","FIDU-Mark","TopLayer","FIDU_Round_Mark_1mm","-15.0000","15.0000","90","Placement Round mark for assembly machine","Standard"
"_XX1","FIDU-Mark","TopLayer","FIDU_Round_Mark_1mm","15.0000","-15.0000","90","Placement Round mark for assembly machine","Standard"
"_XX3","FIDU-Mark","TopLayer","FIDU_Square_Mark_1mm","15.0000","15.0000","90","Placement Square mark for assembly machine","Standard"
```

**Text Output:**
```Pick and Place Locations
C:\PCB_Test\Project Outputs for PCB_Test\Pick Place Output\Pick Place for main.txt

========================================================================================================================
File Design Information:

Date:       22/12/19
Time:       22:19
Revision:   Not in VersionControl
Variant:    No variations
Units used: mm

Designator Comment   Layer       Footprint            Center-X(mm) Center-Y(mm) Rotation Description                                  ComponentKind 
_XX6       FIDU-Mark BottomLayer FIDU_Square_Mark_1mm 10.0000      10.0000      90       "Placement Square mark for assembly machine" Standard      
_XX5       FIDU-Mark BottomLayer FIDU_Round_Mark_1mm  -10.0000     10.0000      90       "Placement Round mark for assembly machine"  Standard      
_XX4       FIDU-Mark BottomLayer FIDU_Round_Mark_1mm  10.0000      -10.0000     90       "Placement Round mark for assembly machine"  Standard      
_XX2       FIDU-Mark TopLayer    FIDU_Round_Mark_1mm  -15.0000     15.0000      90       "Placement Round mark for assembly machine"  Standard      
_XX1       FIDU-Mark TopLayer    FIDU_Round_Mark_1mm  15.0000      -15.0000     90       "Placement Round mark for assembly machine"  Standard      
_XX3       FIDU-Mark TopLayer    FIDU_Square_Mark_1mm 15.0000      15.0000      90       "Placement Square mark for assembly machine" Standard      
```

I hope, this will help you.

p.s.
And this is a setup dialog for Pick&Place export:
[image]

doom-fr, 2019-12-22:
I found this link here :  https://bugs.launchpad.net/pcb/+bug/756972

It can help for orientation.

doom-fr, 2019-12-22:
As you say, I think also there's no standard file format, unless it was csv.
So maybe the best option is to allow the user to enter the file format as a string
for ex : 
"Designator,PositionX,PositionY,Rotation,Side"

In the future, this will be part of the manufacturer configuration.

EdizonTN, 2019-12-22:
> In the future, this will be part of the manufacturer configuration.
- Please note, Pick and place output and BOM are part of ASSEMBLY Configuration, not manufacturer!

...and dont forget: Pick and place files are valid only for SMD devices.

Ex.:
[image]

doom-fr, 2019-12-22:
manufacturer of assembled boards !   ;-)

doom-fr, 2019-12-22:
I think also it will be cool to check if pin 1 is well placed in the footprint.

If not, orientation will be wrong.
But generate the file, warn the user and place a joker character like `*` in the orientation column.

(maybe this orientation check should be also implemented in the library editor).

ubruhin, 2019-12-26:
Thanks for the comments @doom-fr and @EdizonTN!

The initial implementation is ready, maybe one wants to test it: #639

> I think also it will be cool to check if pin 1 is well placed in the footprint. If not, orientation will be wrong.

Actually the rotation doesn't depend on the location of pin 1, but on the orientation of the devices in the tape (or however they are packaged). Thus I think it's not possible for the EDA tool to check if the rotation is correct. Instead, the rotation needs to be checked/corrected in the pick&place software.

EdizonTN, 2019-12-26:
> I think also it will be cool to check if pin 1 is well placed in the footprint.
> 
> If not, orientation will be wrong.
> But generate the file, warn the user and place a joker character like `*` in the orientation column.
> 
> (maybe this orientation check should be also implemented in the library editor).

This check sounds mysteriously for me.
I don't understand, why and how you need check Pin.1 position. Can you describe in more details please (better will be in new thread, I think)?
Thanks

doom-fr, 2019-12-26:
Thanks @ubruhin for this new feature.
Just tested it and it seems to work.
(I will let you know if manufacturers say there is a problem).

But zero is not set to the lower left corner of the board.

ubruhin, 2019-12-26:
> Thanks @ubruhin for this new feature.
> Just tested it and it seems to work.
> (I will let you know if manufacturers say there is a problem).

:+1:

> But zero is not set to the lower left corner of the board.

Zero should be on the same location as in the board editor, so it's up to you whether it is in the lower left corner or not. It's consistent with the Gerber export to avoid offset issues.

doom-fr, 2019-12-27:
Different to https://github.com/LibrePCB/LibrePCB/issues/621#issuecomment-568304775

But make sense.

PR #639 (2019-12-28): Implement pick&place file export
New dialog in the board editor:

[image]

Example output:

```csv
# Pick&Place Position Data File
# 
# Project Name:        Hydro Battery Charger
# Project Version:     v2
# Board Name:          default
# Generation Software: LibrePCB 0.1.4-unstable
# Generation Date:     2019-12-22T21:37:19
# Unit:                mm
# Rotation:            Degrees CCW
# Board Side:          Top

Designator,Value,Device,Package,Position X,Position Y,Rotation,Side
C1,LGW2E471MELZ40 470μF/250V,Elko 22/10mm,Elko 22/10mm,40.005,48.26,180.0,Top
C2,LGW2E471MELZ40 470μF/250V,Elko 22/10 [... cut]
closed 2019-12-28 by ubruhin