LibrePCB/LibrePCB #627: Add LibrePCB to wikipedia

luzpaz, 2019-12-08:
No representation on https://en.wikipedia.org/wiki/Comparison_of_EDA_software

dbrgn, 2019-12-08:
Thanks @doom-fr for adding LibrePCB!

closed 2019-12-08 by dbrgn

doom-fr, 2019-12-08:
Yes, I have done this !
But if someone than knows the project better than me can check what I have written, it might be a good idea.
I have added LibrePCB in the two tables (I don't know exactly why there's two).

And the problem is that we need now to add a page for LibrePCB ... :-)

doom-fr, 2019-12-08:
I just received this : 

> Hello. Your recent edit to Comparison of EDA software appears to have added the name of a non-notable entity to a list that normally includes only notable entries. In general, a person, organization or product added to a list should have a pre-existing article before being added to most lists. If you wish to create such an article, please first confirm that the subject qualifies for a separate, stand-alone article according to Wikipedia's notability guideline. See also WP:COI, in case it applies. 

So the point is not closed ...

I try to add a first version for LibrePCB page.

dbrgn, 2019-12-08:
Cool, thanks! Generally it's considered best-practice on Wikipedia if the authors of a project don't add a WP page themselves, so it's great if you can start something.

The notability guidelines can be found here: https://en.wikipedia.org/wiki/Wikipedia:Notability I'm not sure if LibrePCB qualifies yet, but here is some independent third party coverage that might help, in case it's being disputed:

- Sourcesort interview: https://www.sourcesort.com/interview/urban-bruhin-librepcb
- Adafruit blog picking up the interview: https://blog.adafruit.com/2019/10/31/creating-a-foss-printed-circuit-board-design-tool-opensource-librepcb/
- Blogpost about the new release in a Spanish Linux community blog: https://www.linuxadictos.com/liberada-la-nueva-version-del-software-de-diseno-de-pcb-librepcb-0-1-3.html

The fact that the project received a presentation slot at FOSDEM 2018 and 2020 might also help.

doom-fr, 2019-12-08:
Yes, I am answering that. Thanks for some links I do not know.

doom-fr, 2019-12-08:
Can I also say : 

> LibrePCB output file format is also supported by Aisler, a main PCB manufacturer : https://twitter.com/AislerHQ/status/1203305615096979456?s=20

(I really do not know the size of Aisler)

dbrgn, 2019-12-08:
Yes, that surely helps with regard to notability.