LibrePCB/LibrePCB #628: Request authorization to copy text from LibrePCB website to Wikipedia

doom-fr, 2019-12-09:
##### SUMMARY
I have updated this page of wikipedia : https://en.wikipedia.org/wiki/Comparison_of_EDA_software

And today I receive this message : 

> The material was copied from another website, and thus was a violation of Wikipedia's copyright policy. GNU alone is not a compatible license. Please don't add copyright material to Wikipedia.

##### SOLUTION

Can you please authorize me to copy the texts I need from LibrePCB website to update the above Wikipedia page and create another one for LibrePCB itself ?

If ok, I will link this page to show that you have given me this authorization.

doom-fr, 2019-12-09:
For information, my modifications have been rolled back because I need to first create a page for LibrePCB

ubruhin, 2019-12-12:
Sure, copying text and images from https://librepcb.org/ to Wikipedia is fine for me!

doom-fr, 2020-02-12:
Here is the answer : 

> Doom*, Sorry, but that statement is not adequate for our purposes, because it only allows the material to be copied to Wikipedia. Our license allows the material to be re-copied and reused by anyone for any purpose. So what we need is for the material to be released under a compatible license. There's more information on how to do it at [WP:Donating copyrighted](https://en.wikipedia.org/wiki/Wikipedia:Donating_copyrighted_materials) materials and there's a sample permission email at [WP:consent](https://en.wikipedia.org/wiki/Wikipedia:Declaration_of_consent_for_all_enquiries).

So can you please give to Wikipedia a non-exclusive license under Creative Commons Attribution-Sharealike 3.0 Unported License (CC-BY-SA) and the GNU Free Documentation License (GFDL) (unversioned, with no invariant sections, front-cover texts, or back-cover texts) ?

You can do this with sending an email.

It is explained in the second link.