LibrePCB/LibrePCB #629: Board Editor : Help to build the board with placement highlight + pdf

doom-fr, 2019-12-11:
##### SUMMARY
After getting the PCB from the manufacturer, you have to build the board !
For example, on my board I have near 200 devices and a BOM of 70 lines.
I do by hand the first prototypes.
And I can't just search each time where are the devices on the board.
It wastes too much time. So, some help is needed.

Before, I used to create paper sheets with a board representation and
devices highlighted, one for each BOM line.

##### SOLUTION
1. Allow to highlight devices on the board by BOM line
2. Create a pdf file with board and highlight devices by BOM lines, one line of BOM per page

It will be also interesting to allow ordering this to help the user to follow the implantation rules.

EdizonTN, 2019-12-16:
Yes, this feature is know as "Assembly variants" - one PCB design populated by more than one way.
But this is hard to implement it, I think...

Ex. You create two assembly variant. Each components will have new properties for each variant, called "populated" with boolean value true/false. In schematic are graphically represented by red cross if not populated for current variant.

Additional feature is change component fo each variant.
Ex: component with same footrpint (important!) can be populated by different components (0805 capacitor or 0805 resistor).
As I set, its hard to implement. But when implemented, Value of software increase.

Commonplace is export of assembly data (BOM, Pick&Place, Assmebly print,...) depends on selected variant.

I vote for it.

doom-fr, 2019-12-16:
@EdizonTN , I think that what you are talking about is not the same as the requested feature.

EdizonTN, 2019-12-16:
Ou, sorry, You mean live highlight in PCB Editor and exported PDF.....
hmm, I little bit doing fantasy.

ubruhin, 2025-01-15:
With an interactive HTML BOM being implemented in #1485, I'm not sure whether we really need a fancy assembly PDF generator. Like such a PDF, the HTML is also intended for assisting manual assembly, but is even more powerful. So I'd say this issue can be closed.

Feel free to reopen if this feature is still requested.

doom-fr, 2025-01-22:
The only feature that seems to be missing in #1485 is the ability to order the references to follow the implementation rules (smaller components first, connectors at last, ...).
But this is not a really big problem. Just it will be easier if it exists.

ubruhin, 2025-01-23:
> The only feature that seems to be missing in https://github.com/LibrePCB/LibrePCB/pull/1485 is the ability to order the references to follow the implementation rules (smaller components first, connectors at last, ...).

Thanks for the feedback. If this is the only missing feature, I have good news 😃 I implemented configurable custom ordering which is set to "C, R, L, D, U, Y, X, F" by default. It's only missing in the UI yet, but that will come for sure sooner or later. 

But of course it still doesn't allow do distinguish e.g. between small capacitors and large capacitors...