LibrePCB/LibrePCB #631: Board Editor: Prevent error "Nothing to connect."

mrm2m, 2019-12-15:
##### SUMMARY

If you miss the trace which you want to draw to by just a few mm you will see the message "Nothing to connect".

##### SOLUTION

Find out if there are trace nodes in reasonable distance (like 1cm) around the click position and select the one which is nearest if there is no direct match.

miniwark, 2020-07-20:
I have a similar case of "nothing to connect error", with a handmade component, even if the wire was apparently at the same place than the connection pin. I suspect than the pin was 1 or 2 pixels away from the grid, and the trace try to "snap to grid" instead of "snap to pin".

Also., I have solved the issue by recreating the component from start in the library. Wile doing this, there seem to have a steep, in the Wizard that not editable in the library: In the "Pin-Signal-Map" i have clearly indicated "pin" int the Designator in Schematics" column, and suspect then this steep is important to connect correctly in the schematics (but i do not have find a way to edit this in the library).

5n8ke, 2020-08-11:
Simply choosing the closest NetPoint would not suffice in my opinion. Pads, Vias, and even the closest distance to the traces must be taken into consideration.
And should the currently selected layer for routing be considered as well?

PR #753 (2020-09-01): Board editor: Snap to nearest anchor when starting a trace
When no anchor is found at the start location of drawing a trace, look in a radius of 1cm for other possible candidates.
Currently supports
- NetPoints
- Vias
- Pads

# Principle
The search function looks for anchors in the radius of maxDistance around the specified position on a given layer. When a candidate is found, the radius is reduced to the distance between the position and the center point of the anchor.
The function searches for NetPoints, Vias, and Pads in the given order. It returns the last found anchor since it laid within [... cut]
closed 2020-09-01 by ubruhin