LibrePCB/LibrePCB #642: Pin name differences with Kicad

ouabache, 2019-12-31:
##### SUMMARY
Kicad allows pin names with decimal pts but librePCB does not. This will limit the ability to mass convert kicad libraries to run on librePCB

##### SOLUTION
allow decimal pts in pin names

doom-fr, 2019-12-31:
Is it a really big problem ?
Can't you just convert "." by another character ?

ouabache, 2019-12-31:
It is a big problem and hacking it into another character will only cause
more problems in the future. If an IC vendor puts a "." in a pin name in
the data sheet then they will use that same name in the BSDL file. Somebody
will have to write a program someday to take a board file and bsdl files to
create  an open and shorts test for a board assembly. This becomes a lot
harder when the pin names don't match.

IEEE 1149.1 specifies a character set of A to Z, a to z,0 to 9 and special
characters
" & ' ( ) * , - . : ; < = > _

We should accept any of those

dbrgn, 2019-12-31:
Hmm, IEEE 1149.1 is a standard for electronics testing equipment. How is this relevant for a generic EDA tool?

ouabache, 2019-12-31:
JTAG was created by board test engineers to quickly develop and test
printed circuit boards. If all of your major parts support boundary scan
then it is easy to get a simple open/shorts test running to verify that a
board actually works. JTAG has proven so useful that it is now used
everywhere but it started out as a PCB standard.

Same thing happened with verilog. It was created as a test tool but then IC
designers found out about it and the rest is history.

ouabache, 2020-01-01:
I just ran an experiment where I converted 3500 symbols from the
Kicad,digikey and sparkfun libraries into librePCB. There were 981 failures
due to the use of the following characters:

~ . ( ) []

You could argue that the square brackets are not allowed by the jtag
standard and I would accept that. I do think we should allow the others or
risk having issues in cloning libraries.

I am also seeing a lot of DRC fails due to names not starting with a
capital letter and duplicate pin names. These are commonly  used in the
industry.

ouabache, 2020-01-01:
I take that back. They are not using square brackets.

The librePCB error message was:

File:
/home/johne/LibrePCB-BGA_dev/v0.1/libraries/local/Test.lplib/sym/106caaa5-73b5-4da6-9802-e1658d171709/symbol.lp
Line,Column: -1,-1
Invalid Content: "~RESET"
(../../../../libs/librepcb/common/fileio/sexpression.h:95)
[ WARNING ] Failed to open library element:
"local/Test.lplib/sym/106caaa5-73b5-4da6-9802-e1658d171709" (:0)
[EXCEPTION] Invalid identifier: 'P0[0]'
(../../../../libs/librepcb/common/circuitidentifier.h:54)
[EXCEPTION] File parse error: Invalid identifier: 'P0[0]'

But the file it was parsing did not contain that string. I will update to
the latest and try it again

(librepcb_symbol 106caaa5-73b5-4da6-9802-e1658d171709
 (name "AT45DB161-TC_1")
 (description "created from file---
 ./kicad_memory/AT45DB161-TC_1.csv.\nVariant-default\nGenerated with
librepcb-parts-generator (generate_sym.py)")
 (keywords "   ")
 (author "John E.")
 (version "0.1")
 (created 2019-12-17T00:00:00Z)
 (deprecated false)
 (category b9402ebd-6b2a-4a4b-95ba-7f33873aefa2)
 (pin 999f79bf-9674-4c58-99d7-d63f86d44881 (name "GND")
  (position 10.16 -17.78) (rotation 90.0) (length 2.54)
 )
 (pin de794cd2-6f01-4d0a-b342-9e9a60104528 (name "SI")
  (position -22.86 10.16) (rotation 0.0) (length 2.54)
 )
 (pin e9db0b37-48fd-4fff-93b3-91df1df3c0fd (name "~RESET")
  (position -22.86 7.62) (rotation 0.0) (length 2.54)
 )
 (pin b7d5e6a0-3db2-4df3-8d1c-40b0ab0c5a36 (name "~WP")
  (position -22.86 5.08) (rotation 0.0) (length 2.54)
 )
 (pin 2728651c-3a0b-487c-9870-55d74a4fd13f (name "~CS")
  (position -22.86 2.54) (rotation 0.0) (length 2.54)
 )
 (pin 28f0a72f-32fe-4b3a-b483-8d811c4b4ed6 (name "SCK")
  (position -22.86 0.0) (rotation 0.0) (length 2.54)
 )
 (pin 3dd38b6f-68f0-45e1-a091-e206529d4023 (name "SO")
  (position 22.86 10.16) (rotation 180.0) (length 2.54)
 )
 (pin 7057e8da-ef40-44ce-b009-eb12cb3e9bc6 (name "RDY/~BUSY")
  (position 22.86 7.62) (rotation 180.0) (length 2.54)
 )
 (pin 0a80554e-d091-45bd-9c65-7e01ecc32ce9 (name "VCC")
  (position 5.08 17.78) (rotation 270.0) (length 2.54)
 )
 (polygon 52fa3e03-bd10-4426-a6d3-340a3114f42e (layer sym_outlines)
  (width 0.25) (fill false) (grab_area true)
  (vertex (position -20.32 15.24) (angle 0.0))
  (vertex (position 20.32 15.24) (angle 0.0))
  (vertex (position 20.32 -15.24) (angle 0.0))
  (vertex (position -20.32 -15.24) (angle 0.0))
  (vertex (position -20.32 15.24) (angle 0.0))
 )
 (text 216c6fb2-8b70-442c-93ed-8dd9e5cc4d92 (layer sym_names) (value
"{{NAME}}")
  (align center bottom) (height 2.54) (position -15.24 15.24) (rotation 0.0)
 )
 (text a7cdafb2-1bee-4130-9553-c89dd49b10b4 (layer sym_values) (value
"{{VALUE}}")
  (align center top) (height 2.54) (position -15.24 -15.24) (rotation 0.0)
 )
)

kutis96, 2020-01-24:
> Is it a really big problem ?
> Can't you just convert "." by another character ?

Sure you _can_ do that, but that really doesn't help consistency at all.

Many microcontroller parts, mostly the 8051-derived ones do often specify pin names with dots; say pin P3.6 would correspond to bit 6 of port 3.

Keeping your naming consistent with what's used in the part's documentation and elsewhere is generally a good thing IMO.

PR #1161 (2023-05-07): Allow characters .&() in identifiers
Allow the characters `.`, `&`, `(` and `)` in symbol pin names, package pad names, component signal names, net names and component names (designators). They should not cause any issues within production output files (IPC-D356, Gerber, ...).

There might be more characters useful in some cases, e.g. `:`, `;`, `*`, `%`, `~`, `[` and `]` which are still rejected because some of them could cause problems in production data and others I want to keep reserved for future extensions (e.g. `[]` might be useful for buses, and `~` for overline).

Closes #1074
Closes # [... cut]
closed 2023-05-07 by ubruhin