LibrePCB/LibrePCB #648: Do app updates from within the app

spookydd, 2020-01-04:
Perhaps in the preferences popup in an extra tab, or could be an option in a pull down menu (help maybe): A checkbox to allow automatic search for app updates, and offer updating now if there is one available.

spookydd, 2020-01-05:
I just did an update of libreoffice, which should be similar in many ways to librepcb, at least for its UI behaviors and a few other things.
They do the automatic app updates from within the app and perhaps there is a way to speed things up by looking at how they did it there. Looking into their code. Copying things from one open source project to an other should be totally ok right?

ubruhin, 2020-01-06:
Yeah I think that would be a nice feature, but of course only if LibrePCB was installed with the installer. For other deployment methods (bundle, AppImage, ZIP, AUR, ...) we can't do an automatic update and thus the update feature needs to be disabled.

For the record, the Qt installer framework already supports this use case: LibrePCB just needs to execute `librepcb-maintenance --checkupdates` to check if updates are available. If there are updates, executing `librepcb-maintenance --updater` opens the updater wizard.

But adding the configuration option to workspace settings require to change the settings file format, thus this needs to wait for the 0.2.x release.