LibrePCB/LibrePCB #653: Add manufacturer name field to library elements

dbrgn, 2020-01-09:
### SUMMARY

As discussed in https://librepcb.discourse.group/t/naming-of-packages/167/, the manufacturer field could be included in the library file format for the following entities:

- Symbol
- Component
- Package
- Device

##### SOLUTION

- The field should be optional
- This requires a file format change, so it should probably be part of the v0.2 release
- The field can be used for filtering (e.g. in the "add component" dialog), but it should probably not be shown in the schematics or board editor by default

EdizonTN, 2020-01-10:
If you adding a new field, can you add also some else?
For example, for device:
[image]

dbrgn, 2020-01-12:
While manufacturer name is relevant for the LibrePCB UI itself, I wonder if we could allow arbitrary attributes in the devices, which would act as predefined attributes in the schematics. For example, the manufacturer name could be automatically set as `MANUFACTURER` attribute. A LED device could predefine set the `FORWARD_VOLTAGE` attribute. In the schematics, the user can then always overwrite those attributes if desired.

Order number management could also use this system, some of the attributes like `MPN` or `DIGIKEY` would then be filled in depending on the device variant chosen.

But that's probably a separate feature.

EdizonTN, 2020-01-12:
Yes, I agree. Its a separate feature. 
Have you deadline for v0.2.0 (change file format)?

ubruhin, 2023-07-11:
FYI, added the manufacturer property only to devices (resp. *parts*) as described in #1177. Not sure if it is needed for other library elements as well (as suggested in this issue) so for now I didn't implement that.

closed 2023-07-18 by ubruhin