LibrePCB/LibrePCB #654: Make background color ocnfigurablew

dbrgn, 2020-01-19:
Allow configuring the background color similar to the other board editor colors.

This requires a file format change.

@ubruhin probably an easy fix, right?

ubruhin, 2023-01-15:
@dbrgn I suspect you mean to configure the colors in the workspace settings (not project settings), right? In that case it is even not considered as a file format change since workspace settings are implemented in a generic way.

Actually I wonder if we even should make the whole appearance of schematics and boards configurable in the workspace settings, i.e.:
- Background color of schematics and boards
- The grid style (none/dots/lines) of schematics and boards
- All layer colors of schematics + boards

The grid style and layer colors are currently already stored in projects, I'd remove this configuration then.

My idea is that appearance is a personal preference and has no functional impact at all on the project. The only exception is the grid **interval** which I'd keep in the project settings. Also probably one wants the same appearance for all projects, not just one.

Although this actually **reduces** the flexibility of appearance (per-workspace instead of per-project), IMHO it makes more sense this way. If we ever need per-project settings, we could still extending the implementation of (optional) per-project overrides.

I just wonder if we should allow creating different appearance configurations in a workspace (like themes) which allows to have different styles for dark & light window themes, and could some day be extended to also affect the appearance of the LibrePCB windows (toolbar background, icon sets, ...) :thinking:

dbrgn, 2023-01-16:
Yeah, it definitely makes sense that the non-functional appearance is a user-settings, and not a project-setting.

Themes and a built-in light/dark theme would be cool!

PR #1086 (2023-01-26): Make graphics appearance configurable in workspace settings
Extending workspace settings to allow configuring **background/grid colors, layer colors, grid style** used in all graphics views. Multiple configurations, called *themes* can be created while one of them is used as the active theme. The idea is to support e.g. dark/light themes in future (currently only one builtin theme exists).

[image]

If you like, a schematic could now look like this :smiley: 

[image]

Closes #496, closes #654
closed 2023-01-26 by ubruhin