LibrePCB/LibrePCB #655: Bundled OSX application version is reported as "Created by Qt/QMake"

m4c0, 2020-01-19:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.3
Git Revision:     56bc60d347ff67df0fe1d57807d03f0606de557f
Build Date:       2019-11-30T13:05:08
Qt Version:       5.13.2 (built against 5.13.2)
CPU Architecture: x86_64
Operating System: macOS Mojave (10.14)
Platform Plugin:  cocoa
```

##### SUMMARY

Mac OSX has a feature to "Get Info" about a file - this includes app bundles. For such bundles, it is meant to show, amongst others, the version of the app. For LibrePCB, it currently shows "Created by Qt/QMake"

##### STEPS TO REPRODUCE

1. Open the folder containing the `librepcb.app` in Finder
2. Ctrl-Click (or Right-Click) the app icon
3. Select "Get Info" 

##### EXPECTED RESULTS

The "librepcb Info" box appears, and it reports the application version (0.1.3)

##### ACTUAL RESULTS

The "librepcb Info" box appears, and it reports the application version as "Created by Qt/QMake"

##### ADDITIONAL INFORMATION

From a Mac developer perspective, this is trivial: inside the `app` bundle, there is a file named `Info.plist` inside a folder named `Contents`. That is an XML file describing the bundle. Inside that XML, there is a key-value structure which contains this snippet:

```
<key>CFBundleGetInfoString</key>
<string>Created by Qt/QMake</string>
```

I presume Qt requires some special flag somewhere in its `pro` files to properly mark the version of the app.

spookydd, 2020-10-05:
Hello all,
I just updated to 0.1.5, on OSX (high sierra), and although I thought this had been fixed several months ago, the Get-Info version is reported as "--". Slightly different from earlier, but still not reporting the version.

dbrgn, 2020-10-05:
How did you install LibrePCB, via the installer?

spookydd, 2020-10-05:
Yes, the installer in the dmg.

dbrgn, 2020-10-05:
Ok, in that case someone would need to figure out how that file can be customized when using the [Qt Installer Framework](https://doc.qt.io/qtinstallerframework/index.html).

dbrgn, 2021-01-14:
I think we should wait with this issue until #798 is done. Not sure if that's ready by 0.1.6.

dbrgn, 2021-06-06:
Will get fixed by #798.

closed 2021-09-14 by ubruhin