LibrePCB/LibrePCB #657: Setup: closing Library manager while downloading corrupts libraries system

holgerlembke, 2020-01-23:
##### VERSION / OS / ENVIRONMENT
Windows 10
Workspace on
  \\192.168.xxx.xxx\Developers\Bastelkrams\librepcb workspace

```
Git Revision:     56bc60d347ff67df0fe1d57807d03f0606de557f
Build Date:       2019-11-30T13:03:21
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Windows 10 (10.0)
Platform Plugin:  windows
...
```

##### SUMMARY
Closing Library Manager while downloading corrupts the system.

Repository list does not show up any more. Placing parts results in "not found errors".

##### STEPS TO REPRODUCE
First time user. After  downloading followed all the steps, ending in the Library Manager, clicked "[] select all" and "download" and after seeing the downloads starting, clicked on "close".

I then deleted the workspace and added the libraries again and did not close the dialog until everything was downloaded. Works now.

Seems to me that "closing while download active" should be blocked.

holgerlembke, 2020-01-23:
Addendum: Access to \\\192.168.xxx.xxx\ is quite slow. 

I had to learn to wait for "scanning libraries" to finish so that changes to libs are visible/active. For a starter it is a little bit confusing that I can continue working but don't see changes, until scan is done.

I assume that a lot of work is still to be done on the lib thing... so I won't gibe smart advices like "don't rescan all libs when only one has been changed". ;-)

ubruhin, 2020-01-27:
Hi,

I tried to reproduce this, but did not found a case where something gets corrupted... Are you sure it was necessary to delete the workspace? And where did you get the "not found errors" exactly? Maybe a screenshot would help.

> I assume that a lot of work is still to be done on the lib thing... so I won't give smart advises like "don't rescan all libs when only one has been changed". ;-)

Yes we know the library scan is currently not very efficient, there are plans to improve that. But it's not high priority since usually it's not a big issue since the scan is performed in the background and "only" takes ~10-30s ;)

holgerlembke, 2020-01-29:
In some weeks time I will have access to a reset-everything-on-reboot-PC and a long lunch break. I added this bug to my work list to try to replicate it.

holgerlembke, 2020-02-26:
Ok, here is how to reproduce the workspace problems during install. I made three test runs on Windows 10, everytime cloned untouched system with LibrePCB installed with standard steps, nothing fancy.

-- welcome to librepcb -> continue
-- selet workspace path/create new workspace -> continue
-- workspace settings/user name -> finish

-- "click here to open library manager"
-- [x] select all
-- click "download and install/update selected libraries"

And at this step, fast close all open windows with X-button. It can be done slower, if the target destination is slow (this is my case, network share), on a i7/SSD/fast internet I need to be very fast. I ended at random problems, see screen shot, that some downloads seem to be in a not recognized state or workspace won't load. Sometime I can't open the created workspace, sometimes it is fixed by creating a new one. 

To me it looks like it depends in which state the process is interrupted....

I'm just surprised that I can close dialogs while a process is still running that seems to require that dialog.

[image]

ubruhin, 2020-09-26:
Thanks for the detailed report @holgerlembke! Now with the exact error message it was easy to found the issue. The download simply fails if the (temporary) ZIP file in the workspace already exists. I suspect, everything should work fine after deleting that ZIP file. I'll fix this by explicitly deleting the ZIP file before starting a new download. Then it should no longer be a problem when closing LibrePCB while a download is still in progress.

PR #775 (2020-09-29): LibraryDownload: Remove destination ZIP before starting download
Affects downloading libraries with the library manager:

If the temporary ZIP file already exists before starting to download a library, the download failed with a "file already exists" error. This can happen for example when closing LibrePCB while a library download was in progress. Afterwards, it was not possible anymore to download the library with the library manager.

Now the ZIP file will be removed before starting the download.

Fixes #657.
closed 2020-09-29 by ubruhin