LibrePCB/LibrePCB #659: add other licenses

cjdg, 2020-01-26:
it should be possible to add more licenses on new proyects

dbrgn, 2020-01-26:
Which licenses would you suggest?

My suggestions would be:

- TAPR OHL
- CERN OHL 1.2

CERN OHL 2.0 is also [very promising](https://www.youtube.com/watch?v=Lg0BcLdfV-I), but still in beta.

(Note that you can *always* add other licenses as you wish, the wizard simply suggests some common ones.)

cjdg, 2020-02-02:
Can be both? or if there are more would be better, maybe making a repo for libraries, include some, and allows the sofotware to add manually from server or localy?

dbrgn, 2020-02-02:
I don't understand your question. This is only about the licenses that LibrePCB suggests in the license picker.

You can always add your own licenses to your projects or libraries by adding the necessary license files.

cjdg, 2020-02-02:
Yes but its ideal for final users to have at hand the licenses, remember, not all the users know all the licenses, and would be nice so they can have a little bit more options

dbrgn, 2020-02-03:
Sure, I'm all for adding more options :slightly_smiling_face: However, before licensing your project under a license, you have to understand that license. This is a step that LibrePCB cannot solve for you. We can just help you to apply a license once you've picked one.

I'll add a PR with TAPR OHL and CERN OHL.

closed 2024-03-28 by ubruhin