LibrePCB/LibrePCB #660: Add bar above/below signal name for active low/high pins

sixtyfive, 2020-01-26:
KiCad has a convention of one (`~`) or two (`~~`) tildes to put a bar above or below a signal name:

> Rather than selecting the active low graphical pin style, active low pins should be designated using a bar above the pin name.
>
> This provides a cleaner look for the schematic.
> 
> To place a line above the pin text, prefix the name with the tilde (~) character. In case of a pin name where only part of it is active low, and hence, not all of it needs a bar above, two tilde characters must be used as delimitation, e.g. ~SPD_LED~/GPIO2.
> 
> If the pin name given by the manufacturer includes an active low prefix or suffix, it should be removed when adding the bar above it to avoid negating it twice. For example for nRESET, use ~RESET.

Example usage from a TI datasheet:

[image]

Any chance of something similar to this for LibrePCB?

dbrgn, 2020-01-26:
Another one would be `nRESET` / `!RESET` / `~RESET`.

sixtyfive, 2020-01-26:
`nRESET` is also mentioned in the above quote from the KiCad docs. I don't think it matters which notation is used, be it `n` or `!` or `~`, as long as there is some way of getting a bar above or under a signal name somehow.

dbrgn, 2020-01-26:
Yep. I think right now the issue is that "bar above words" is not supported out-of-the-box by Qt's text label component, so we'd probably need to create a custom widget for that. And we'd have to bikeshed the syntax.

sixtyfive, 2020-01-26:
Does [U+203E ‾ OVERLINE](https://en.wikipedia.org/wiki/Overline#Unicode) work there?

dbrgn, 2020-01-26:
Hm, good idea, there's "U+0305 Combining Overline". However, it seems to be a bit buggy depending on the renderer.

Text: `R̅E̅S̅E̅T`

Terminal (alacritty):

[image]

Firefox Textarea:

[image]

Firefox URL bar:

[image]

sixtyfive, 2020-01-26:
Hmm, looks as if it might have been designed to draw Devanagari instead of Latin script, since all three renderers get it wrong. Shame :-/

dbrgn, 2020-01-28:
I took a quick look at implementation options. When rendering a text element in Qt, an overline can easily be added with `setOverline(true)`, but the thing is that it only works for the entire text element, not for parts of it.

@ubruhin can font / styles be applied partially in a text element?

sixtyfive, 2020-01-28:
I've never seen pins with an overline across only part of the name, except perhaps if there's something like `1 (R̅E̅S̅E̅T̅)`, which doesn't seem to be customary in LibrePCB, or does it happen?

dbrgn, 2020-01-28:
It is being used sometimes when pins have alternate functions. E.g. on the ATtiny85:

[image]

EdizonTN, 2020-01-28:
Hi all, 
my 2 cents:
- I know one unnamned commercial EDA, use to create a bar above character - an backslash symbol
[image]

- And yes, pins with an overline across only part of the nameis absolute normal for todays. Have a look to microcontroller pin names. One pin can shared for more functions. It's normal. And for netnames too, of course.

[image]

baryluk, 2020-10-05:
`!{RESET}/PIO0_5`. Just an idea. It also allows for each multiple negated and non-negated designations on the same pin. Easy to check automatically for correctness too.

sixtyfive, 2020-10-14:
Much nicer than the backslash-orgy.

ubruhin, 2023-06-14:
I looked into this. Unfortunately, partially overlined text is indeed not supported by `QPainter` which we use for rendering. But I think I have found a way to determine the overline coordinates to draw them manually.

Anyway, the main question is how the overlines are specified in the pin/net names. Here my thoughts about the suggestions already posted in this issue:

- `~RESET`: Not bad, but I already saw several schematics where the symbol `~` is contained in net names to indicate AC (e.g. `VIN~` or `12V~`). So users *might* be confused if `~` is rendered as an overline. Also the need for `~~` to toggle overline off is a bit verbose.
- `!{RESET}`: A bit verbose in my opinion, and `{}` might cause issues in some export formats.
- `\R\E\S\E\T`: Too cumbersome to type ;-)

So I ended up with this proposal for concrete rules:
1. A single `!` toggles the overline on/off, depending on its previous state (initial state is off).
2. The character `/` implicitly switches off the overline before rendering. This can be prevented by prefixing it with `!`.
3. A double `!!` has no effect on overlines, they are rendered as a single `!`. In case of an odd number of `!` (e.g. `!!!`), the last one toggles overline on/off.
4. Important: All `!` at the end of the string have no effect, they are rendered as-is (bypassing rules 1 and 3).

Advantages:
- `!` is intuitive to express inversion
- `!` is basically never needed in net names, so it's pretty safe to make it functional
- `!` is safe to be exported e.g. in netlists (where overline rendering is not possible)
- Not verbose at all (not even needed to toggle off overline in typical use-cases)
- Still flexible to cover unusual use-cases (e.g. `!!` to render it, or avoid auto-off by additional `!`). The only impossible thing is to toggle overline off and render a `!` as the next character.
- Guaranteed unique rendering because only `!` which do have an effect are consumed, any others are rendered as-is (avoids identical rendering of two different net names like `RESET` and `RESET!`).

Examples:

| Input | Rendering |
|---|---|
| `!RESET/SHDN` | `R̅E̅S̅E̅T̅/SHDN` |
| `RESET/!SHDN` | `RESET/S̅H̅D̅N̅` |
| `!D+` | `D̅+̅ ` |
| `!A!/B` | `A̅/̅B̅` |
| `!A/!B` | `A̅/B̅` |
| `!ABC!DEF` | `A̅B̅C̅DEF` |
| `DANGER!!RTFM` | `DANGER!RTFM` |
| `DANGER!` | `DANGER!` |
| `DANGER!!` | `DANGER!!` |

PR #1172 (2023-06-19): Render '!' in pin-, pad- and net names as overlines
Support rendering overlines by adding `!` to pin-, pad-, or net names. The exact rules as described in https://github.com/LibrePCB/LibrePCB/issues/660#issuecomment-1590891734:

1. A single `!` toggles the overline on/off, depending on its previous state (initial state is off).
2. The character `/` implicitly switches off the overline before rendering. This can be prevented by prefixing it with `!`.
3. A double `!!` has no effect on overlines, they are rendered as a single `!`. In case of an odd number of `!` (e.g. `!!!`), the **last** one tog [... cut]