LibrePCB/LibrePCB #661: Unclear on how to contribute parts

sixtyfive, 2020-01-26:
So for simplicity's sake (and to learn) I began a library called "workshop", which now already has some useful generic parts (like generic 3mm and 5mm diameter LEDs) and one slightly bigger IC that might also be interesting for others (the 1970s ADC0804). I'm wondering how to go about contributing these, or whether, perhaps, it's still too early in the development of the LibrePCB library system to do so:

- it seems useful to keep such a "workshop" library around for personal use, but I can find neither a "copy-paste" feature nor a "export the selected items" feature; so how does one go about freeing those symbols/components/devices that seem worthy for contribution from such a personal mixed-bag-library?

- in the case of the '0804, it was originally manufactured by _NatSemi_, then later also by _intersil_ and _Philips_, and is now still being produced by _TI_. The signal names are the same, the packages are the same, even the absolute maximum ratings are, I think, the same. Of these manufacturers, I can only see an existing LibrePCB library for _TI_. Should three more libraries be created and then the part be submitted four times in total? Or does it belong into the _LibrePCB Integrated Circuits_ library instead?

dbrgn, 2020-01-26:
Hi! We definitely need a formal contribution guide. I started a guide here: https://librepcb.discourse.group/t/how-to-contribute-to-libraries/140/3?u=dbrgn

> but I can find neither a "copy-paste" feature nor a "export the selected items" feature

To move parts from one library to another, the current master version contains a tool to do exactly that: #638 However, that's not yet in the stable version. So right now you need to copy the directory in the file system.

Regarding your other question with the ADC0804, I'm not actually sure. @ubruhin what do you think?

dbrgn, 2020-01-26:
Btw, for user questions, the Forum (https://librepcb.discourse.group/) is probably the best place to ask.

ubruhin, 2023-02-21:
As this issue is not related to the source code of LibrePCB and has already been picked up in the forum (and in the mean time is even mentioned [on the website](https://librepcb.org/contribute/)) I'll close this issue here. Also the mentioned move/copy-to-other-library feature is now available in the latest stable release.

closed 2023-02-21 by ubruhin