LibrePCB/LibrePCB #663: Board Editor: select part

holgerlembke, 2020-01-29:
##### SUMMARY
Add some more ways to select a part.

So I added some parts to my pcb and made a placement error. Now I can't move the part because I can't select it.

[image]

##### SOLUTION
I would expect to have some click actions: first click on pad selects pad, next click (double click?) selects entire part for movement. Additional a context menu could be nice to lock parts at position just to avoid errors.

I miss a "this are your parts on the board"-list where i can click on the part to select it (like the panel to add not placed parts).

SchnWalter, 2020-01-29:
You can drag a part by its "center point", you can check it in the library package editor, it's the the 0,0 point.

holgerlembke, 2020-01-29:
I know. 

But where in my screenshot is this center point? It is highly impractical to look up the center point... and if found, what if it is under a pad or near a pad?

I would suggest a more user friendly solution.

doom-fr, 2020-01-29:
@holgerlembke can you put a screenshot of activated layers ?
Some need to be visible to allow dragging components (Top reference I think).
And also, it will be better if you have something in Top Documentation layer.

doom-fr, 2020-01-29:
This issue https://github.com/LibrePCB/LibrePCB/issues/536 might be related

holgerlembke, 2020-01-29:
@doom-fr Layers? I don't find anything with layers in board editor I could select.... (other than when placing parts/traces).

Parts are all on layer "top copper".

If I move away some other parts I can select the part. But in that situation, I found no way.

doom-fr, 2020-01-29:
On the right I have something like this 
[image]

If you don't have it, click View > Layers in the board editor.

doom-fr, 2020-01-29:
You have also to select the tab Layers after adding it.
(@ubruhin, I think that this should be brought to front when going from 'not view' to 'view').

holgerlembke, 2020-01-30:
/me noob. :-)

Found it.

It helps. I turned off the pads-layer and by guess-and-try used some center crosses to move something until I found the right cross.

This is not a nice ways....

doom-fr, 2020-01-30:
No problem, I was also noob some months ago.

I am pretty sure it will be easier if you draw the outline of the package according to documentation.
But yes, it can be enhanced.

ubruhin, 2026-01-03:
Should be fixed by #1533.