LibrePCB/LibrePCB #664: Ship documentation with source .zip archive

FedericoCeratto, 2020-01-31:
Hello and thanks for developing LibrePCB.
Can you please ship a copy of the documentation in the source .zip archive? It would allow installing the docs locally by Linux distributions.
It could be in AsciiDoc format.

ubruhin, 2020-02-08:
Hi,

Couldn't the documentation be downloaded from https://github.com/LibrePCB/librepcb-doc directly? Somehow I don't like to add it to the sources .zip archive since documentation can get quite large with all the images etc.

And wouldn't be HTML better than Asciidoc files? In future we might provide a separate .zip archives with the HTML docs.

FedericoCeratto, 2020-02-08:
It's quite common for upstream projects to ship documentation markdown/rst/AsciiDoc files as part of the source tarball/zip in order to ensure that the documentation is consistent with features, UI and UX of the released software.
Distributions can then generate HTML files as part of the build process and treat them as a build artifact to be installed with its own "binary" package.
(This can also be required to satisfy some licenses where HTML/PDFs are not considered the "preferred form" for editing)
Anyhow, if librepcb-doc evolves independently from LibrePCB (aka they are not released in lockstep) it's not a problem and it can be packaged independently.

ubruhin, 2023-02-21:
> Anyhow, if librepcb-doc evolves independently from LibrePCB (aka they are not released in lockstep) it's not a problem and it can be packaged independently.

This is actually the case - I recommend to just package the latest snapshot of `librepcb-doc` at the time LibrePCB is packaged. 

If I understand you correctly, that workflow good enough so there's nothing we need to change at our side, i.e. this issue can be closed(?). If not, please feel free to reopen :)

closed 2023-02-21 by ubruhin

FedericoCeratto, 2023-02-21:
@ubruhin packaging the documentation as a different project is very uncommon but still workable, thank you :)
[If you were to git-tag the documentation repo with the same version number as LibrePCB it could help prevent confusion for end users and other distros]

ubruhin, 2023-02-21:
> packaging the documentation as a different project is very uncommon

As far as I can see, it's uncommon for non-GUI tools - which makes perfectly sense since such a documentation consists only of text and is thus very lightweight. However, I am not aware of (larger) GUI applications shipping the docs together with the sources. I think the reason is simple: A GUI documentation consists of a huge amount of screenshots (and maybe even videos) so the docs would be way too heavy (can be even hundrets of megabytes). In my opinion, this is way too bloated for shipping it together with the source code. If we would do, I'm 100% sure people will complain about the large source ZIP.

Anyway, since separate distribution is fine as well, no further discussion is needed about that :see_no_evil: 

> [If you were to git-tag the documentation repo with the same version number as LibrePCB it could help prevent confusion for end users and other distros]

Since the docs and the application are developed independently, this is not really possible (there is not only one docs version corresponding to the application version). Also at the time a new application release is made, the docs are not updated yet - it may take months until they get updated to the latest application version.

Of course it would be much better to release application and docs together, but unfortunately we don't have resources for that. There are no developers payed for working on the docs :wink:

FedericoCeratto, 2023-02-21:
Works for me, I'll keep an eye on the doc repo! Thanks again for the answers!