LibrePCB/LibrePCB #667: Routing layer automatically changes from user selection when routing between vias

sixtyfive, 2020-02-07:
This might be related to #614.

When a net already has a trace on e.g. the top copper layer and now two vias for that same net are added, the expectation would be that connections between those two vias can also be made on another, e.g. the bottom copper layer.

Instead, after selecting "Bot Copper" as the layer for the trace tool, and clicking on any one of the vias, the layer automatically reverts back to "Top Copper" (or whatever layer the net in question already has traces on) and another track is created on that layer.

doom-fr, 2020-02-07:
Are you sure that the vias are attached to the wire ?
See the linked issue for needed tests.
D.

5n8ke, 2020-08-07:
This is fixed by #733 (at least could not be reproduced)

closed 2020-08-31 by 5n8ke