LibrePCB/LibrePCB #676: Can't "place devices" macOS

sfcgeorge, 2020-03-08:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.3
Git Revision:     56bc60d347ff67df0fe1d57807d03f0606de557f
Build Date:       2019-11-30T13:05:08
Qt Version:       5.13.2 (built against 5.13.2)
CPU Architecture: x86_64
Operating System: macOS 10.15
Platform Plugin:  cocoa
```

##### SUMMARY

Doesn't seem possible to place devices, I see the list of devices but have no way of placing them.

[image]

##### STEPS TO REPRODUCE

* Place some components in Schematic Editor, ignore various warnings you don't understand.
* Switch to Board Editor, helpful banner says components are not placed so go to Place Devices dock.
* There is a list of devices but clicking on them the "Device:" label is blank and the "Footprint:" dropdown does nothing - can't place devices.

##### ADDITIONAL INFORMATION

Not sure if this is a bug or I just skipped some steps. If I have missed something then it would be helpful if the UI said what else I have to do.

Still, it seems way clearer and better designed than Eagle or KiCad, those have entirely stumped me.

ubruhin, 2020-03-08:
Hi,

Hmm that's strange. The problem might be that there is simply no device for those components available in our libraries, but actually in that case the board editor should show a message why no device can be chosen.

Could you add a more common component like a resistor to your schematic to check if the "add device" feature is working properly if there are any devices found?

sfcgeorge, 2020-03-08:
Huh, you're spot on, thanks for the quick feedback. I'll close this issue as the real issue is the built in component library doesn't have many corresponding footprints. 

Since I tried with 4 different components I didn't think that could be the issue but evidently I got unlucky; things like resistors and capacitors are fine.

Other than expanding the component library, a nice improvement could be placeholder text saying "This component doesn't have a corresponding PCB footprint."

ubruhin, 2020-03-08:
OK, good to hear it's actually working :)

> Other than expanding the component library, a nice improvement could be placeholder text saying "This component doesn't have a corresponding PCB footprint."

Actually that's the case, see this screenshot:

[image]

But it seems for some reason this text was not displayed in your case, and I have no idea why... :(

sfcgeorge, 2020-03-10:
Oh cool, yeah that would've been perfect. That looks like a custom UI element so maybe it hasn't been ported to Mac. Or as the text is rather long, perhaps on the Mac rather than overflowing it just disappears. Just guesses as I don't know QT.

dbrgn, 2020-03-10:
I'll try to reproduce the issue in the office, where I have access to a macOS device.

Or maybe it could be tested using a UI test + CI? @ubruhin

ubruhin, 2020-04-13:
> I'll try to reproduce the issue in the office, where I have access to a macOS device.

Did you test it in the meantime? :slightly_smiling_face: 

> Or maybe it could be tested using a UI test + CI? @ubruhin

I think the only thing we could do (with reasonable effort) is to check whether the "visible" property of the label is set properly, but I don't expect it to detect this problem since the logic works correctly on other operating systems. Maybe it's an issue with stylesheets or something like that...

dbrgn, 2020-04-13:
> Did you test it in the meantime?

No, unfortunately with the current COVID situation I'm not in the office anymore and don't have access to a macOS device.

spookydd, 2020-04-13:
> On Apr 13, 2020, at 11:06 PM, Danilo Bargen <notifications@github.com> wrote:
> 
> Did you test it in the meantime?
> 
> No, unfortunately with the current COVID situation I’m not in the office anymore and don't have access to a macOS device.

I’m on Mac. I could run some tests.

I’m not up to date on what’s going on with the project and I don’t have the most recent version, but if I’m pointed the version to test, I’ll grab it.

And someone could explain what to test for, so I can do the right thing..

-- 
Didier Godefroy
dg@ulysium.net
Support Anti-Spam legislation
Join the fight http://www.cauce.org

dbrgn, 2020-04-13:
@spookydd steps to reproduce:

1. On macOS, open LibrePCB with a new project
2. In the schematics, place a component that does not have any corresponding device
3. Switch to the PCB view, go to device placement, select that component

If everything works the way it should, you should now see a red warning message as shown in the screenshot in https://github.com/LibrePCB/LibrePCB/issues/676#issuecomment-596254738.

spookydd, 2020-04-14:
Which package should I grab?

I have a previously installed version from several weeks ago, which would need to be replaced by one more recent.

I need to know what to grab, where to get that right package. I’m not using homebrew, it needs to be a precompiled drop-in replacement for that earlier installed one.

Is there a specific component already there that I could use to test this? Or do I need to make one up?

> 
> 
> @spookydd steps to reproduce:
> 
> 	• On macOS, open LibrePCB with a new project
> 	• In the schematics, place a component that does not have any corresponding device
> 	• Switch to the PCB view, go to device placement, select that component
> If everything works the way it should, you should now see a red warning message as shown in the screenshot in #676 (comment).

dbrgn, 2020-04-14:
> I have a previously installed version from several weeks ago, which would need to be replaced by one more recent.

Testing on the last released version (0.1.3) is sufficient. There hasn't been a fix for this so far, the question is whether the problem always occurs on macOS or only in certain configurations.

> Is there a specific component already there that I could use to test this? Or do I need to make one up?

If you add a new component to the schematic, in the list of components, you will see an arrow to the left of the components that have available devices.

[image]

Here, "Transistor NPN" or "MOSFET P-Channel Dual" do not have devices available, so you could pick one of those.

spookydd, 2020-04-16:
> On Apr 14, 2020, at 9:50 AM, Danilo Bargen <notifications@github.com> wrote:
> 
> I have a previously installed version from several weeks ago, which would need to be replaced by one more recent.
> 
> Testing on the last released version (0.1.3) is sufficient. There hasn’t been a fix for this so far, the question is whether the problem always occurs on macOS or only in certain configurations.

The version I have may be older that thought, and isn’t showing its version in the info popup. That issue I think got fixed maybe more recently, so I think I need to get a newer one.

I couldn’t find where to grab a more recent one that is precompiled and as an installer, not a homebrew install.

Where can I find this please?

Thanks,

-- 
Didier Godefroy
dg@ulysium.net
Support Anti-Spam legislation
Join the fight http://www.cauce.org

spookydd, 2020-04-17:
Hi all,

> On Apr 16, 2020, at 11:01 PM, Danilo Bargen <notifications@github.com> wrote:
> 
> Check out the website: https://librepcb.org/download/

I didn’t think of that. I thought maybe a latest work in progress version was better to use..

Anyway I grabbed what was there, which is v0.1.3 and it turns out what I was running was more recent than that one, the v0.1.4-unstable, which I tested things on before.

I thought the issue with the version in the mac’s get-info popup had been fixed, but it’s not.

I get this instead of the version when I do the get-info “Created by Qt/QMake”.

Not very helpful and useless info.

One other thing I thought was also fixed, was the default install folder that the installer uses, which should be the main system Application and not the one located in the user’s root folder.

Anyway, after running it and double checking on the version, which isn’t shown in get-info, it is indeed v0.1.3, but I had to override the default install location to prevent it from going into the Applications folder in the user’s root.

Now I made up a new test project and added an NPN part as mentioned (see attached window screenshot), and then switched to the board design window and see attached…

I’m not sure if I may be missing a step or something, but that’s what I get right from switching from the schematic..

I was expecting a bit more… Am I missing something?

-- 
Didier Godefroy
dg@ulysium.net
Support Anti-Spam legislation
Join the fight http://www.cauce.org

dbrgn, 2020-04-17:
> I thought the issue with the version in the mac’s get-info popup had been fixed, but it’s not.

No, that issue is still open: https://github.com/LibrePCB/LibrePCB/issues/655 But for now you can open the "about" screen in LibrePCB itself, which provides much more detailed information about your currently installed version.

> One other thing I thought was also fixed, was the default install folder that the installer uses, which should be the main system Application and not the one located in the user’s root folder.

That is fixed in the current nightly version, but it's not yet released. See https://github.com/LibrePCB/LibrePCB/pull/649

> and see attached

We cannot see your attachment. Could you maybe upload the pictures through the GitHub website?

spookydd, 2020-04-18:
> On Apr 18, 2020, at 1:13 AM, Danilo Bargen <notifications@github.com> wrote:
> 
> I thought the issue with the version in the mac’s get-info popup had been fixed, but it’s not.
> 
> No, that issue is still open: #655 But for now you can open the “about" screen in LibrePCB itself, which provides much more detailed information about your currently installed version.

Ok, then I jumped the gun…

> One other thing I thought was also fixed, was the default install folder that the installer uses, which should be the main system Application and not the one located in the user’s root folder.
> 
> That is fixed in the current nightly version, but it's not yet released. See #649

Will that be in 0.1.4 ?

> and see attached
> 
> We cannot see your attachment. Could you maybe upload the pictures through the GitHub website?

I sure can, if you point me to the right place and how to do so. I’m not familiar at all with GitHub, so I need to be guided there.

Whichever way will be fine, but for GitHub I need info.
Those screenshots are regular Mac png files, so should be no trouble.
I wonder why they’re not showing up in my messages..
One other way would be for me to send to someone’s email..

-- 
Didier Godefroy
dg@ulysium.net
Support Anti-Spam legislation
Join the fight http://www.cauce.org

dbrgn, 2020-04-18:
> Will that be in 0.1.4 ?

Yes.

> I sure can, if you point me to the right place and how to do so. I’m not familiar at all with GitHub, so I need to be guided there.

Just go to https://github.com/LibrePCB/LibrePCB/issues/676 and drag&drop the images into the comment field. The email-to-comment system of GitHub only handles text, not attachments.

spookydd, 2020-04-18:
Ok, posting those 2 screenshots.
I'm on High Sierra 10.13.6 and v0.1.3
[image]
[image]

dbrgn, 2020-04-18:
Ok, thanks! In that case it's definitely still a problem and should be considered a bug.

alexantao, 2022-11-13:
Hi.  Let me resuscitate this issue.  
I’m very new on all this stuff, never designed a PCB, etc, but I’m learning and wanted to make my very first And simple PCB for an Arduino project.
I made the schematics but when I tried to place them on the board, things like a potentiometer did not have the footprint.  
I understand that an Arduino  depends on external hands (it has the schematics), but simple things like a potentiometer should not be missing. Or am I doing something wrong? 
I have librepcb app image on Ubuntu.  Last version and all libraries downloaded.  Thanks

alexantao, 2022-11-13:
Hi.  Let me resuscitate this issue.  
I’m very new on all this stuff, never designed a PCB, etc, but I’m learning and wanted to make my very first And simple PCB for an Arduino project.
I made the schematics but when I tried to place them on the board, things like a potentiometer did not have the footprint.  
I understand that an Arduino  depends on external hands (it has the schematics), but simple things like a potentiometer should not be missing. Is this the same bug Or am I doing something wrong? 
I have librepcb app image on Ubuntu.  Last version and all libraries downloaded.  Thanks

ubruhin, 2022-11-21:
@alexantao No, it's not a problem of the application. See https://github.com/LibrePCB/LibrePCB/issues/1064#issuecomment-1321982127 for explanation.

ubruhin, 2023-04-15:
I was now able to test this on a mac. With LibrePCB 0.1.7 everything works as expected (warning message is shown). So I'll close this issue. Let me know if someone still experiences this problem with 0.1.7.

closed 2023-04-15 by ubruhin