LibrePCB/LibrePCB #679: Device editor: Assigning many pads to signals is inefficient

ubruhin, 2020-03-23:
##### SUMMARY
For devices with many pads, it's a pain to connect all of them to their corresponding component signals: open drop-down, choose signal, move focus to next dropdown, open dropdown, choose signal, ...

##### SOLUTION
Would be nice to have a more efficient way to assign pads. Maybe let LibrePCB iterate somehow trough all pads, and let the user choose the corresponding signal with a single click. Then it would need only 1 click per signal, instead of 3 clicks.

For components with many signals it might be even more helpful if there is a search field to search for component signals instead of scrolling through a long list of signals.

But no idea how *exactly* the corresponding GUI could look like :wink:

ouabache, 2020-03-24:
The only way to handle large pin count devices is to enter the
PIN_NAME,PAD_NUMBER in a machine readable form and use a script to generate
the symbol,component and device files. I have a python script that can do
this with FPGAs in csv file format.

doom-fr, 2020-03-31:
And also hide already assigned signals.

As usually there's 1 pad for 1 signal, it will be easier to found the remaining signal in the list.

And as user also needs sometimes to reassign a signal, add a checkbox or something else to open the full list (or just the list of already assigned signals).

It will transform a n² problem to a n²/2 problem !

khenriks, 2021-04-16:
In cases where the pad name and signal name are the same (like say both are 1..20), it would be really nice to have a button that would just auto-associate them, so we don't have to pick them at all. That would have really helped me with a couple devices.

PR #1309 (2024-02-23): Device editor: Add tools to help defining the pinout
Adding some tools to the device editor to assign pads to signals in much more efficient ways:

- **Auto-connection tool** to assign pads to signals by matching names. Only exact matches (but case-insensitive) are taken into account, no magic going on at all.
- **Interactive tool** to connect any remaining (i.e. unconnected) pad one by one, by providing a list of signals which is sorted in a sophisticated way (numerically, best matches at top, already connected signals at bottom). In addition, the list can be filtered by typing. The interacti [... cut]
closed 2024-02-23 by ubruhin