LibrePCB/LibrePCB #681: Schematic Editor : last page of schematics disappears

doom-fr, 2020-03-31:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.4-unstable
Git Revision:     16bdf52a7aabdb4830a771d911f89d0c79c5a374
Build Date:       2020-01-08T19:13:29
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.4 LTS
Platform Plugin:  xcb
```

and also in last nightly build

```
LibrePCB Version: 0.2.0-unstable
Git Revision:     22960e3f2410972903ee06a7f5ae99c2faa88734
Build Date:       2020-03-28T11:32:59
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.4 LTS
Platform Plugin:  xcb
```

##### SUMMARY
I open a project and one page of schematic is missing.
It was the last one

##### STEPS TO REPRODUCE
1. open a project
2. look at the list of page

##### EXPECTED RESULTS
3 pages

##### ACTUAL RESULTS
2 pages
[image]

##### ADDITIONAL INFORMATION
The files seems to be there : 
[image]

content of `schematics/schematic.lp`
```
(librepcb_schematics
 (schematic "schematics/schematic/schematic.lp")
 (schematic "schematics/schematic_p2/schematic.lp")
 (schematic "schematics/schematic_p3/schematic.lp")
)
```

doom-fr, 2020-03-31:
```
[DEBUG-MSG] load project library... (:0)
[DEBUG-MSG] successfully loaded 56 symbols (:0)
[DEBUG-MSG] successfully loaded 45 packages (:0)
[DEBUG-MSG] successfully loaded 46 components (:0)
[DEBUG-MSG] successfully loaded 53 devices (:0)
[DEBUG-MSG] project library successfully loaded! (:0)
[DEBUG-MSG] load circuit... (:0)
[DEBUG-MSG] circuit successfully loaded! (:0)
[DEBUG-MSG] 3 schematics successfully loaded! (:0)
```

3 schematics !!!

What I also want to add is that I have fully reinstalled my PC (ubuntu 18.04 LTS) some weeks ago.
But I have all files !

doom-fr, 2020-03-31:
And if I add a new schematic page, it appears back !
but new save does not change the problem

doom-fr, 2020-03-31:
Found !
In fact, there was a problem of display of the previews (miniatures).
If you change the size of the column, it appears back, and then all is ok.

ubruhin, 2020-04-07:
Indeed, I was able to reproduce this issue! It seems that sometimes pages are wrapped into a second column and might not be visible at all, depending on the dock width:

[image]

PR #685 (2020-04-14): SchematicPagesDock: Disable wrapping of schematic pages
If there is not enough space available to display the schematic pages in a vertical layout, QListWidget was allowed to wrap pages to the right, i.e. using a horizontal layout. In some situations (depending on the width of the dock widget), the wrapped pages were not visible anymore, so it looked like some schematic pages disappeared. By disabling wrapping, QListWidget always uses a vertical layout.

Fixes #681
closed 2020-04-14 by ubruhin