LibrePCB/LibrePCB #701: Library & Schematic editor : allow to modify lines & polygones with graphical method

doom-fr, 2020-05-17:
##### SUMMARY
When you draw a line or a polygone, it is with graphical method.
But when you want to change it, you can only edit it with raw coordinates

##### SOLUTION
Allow to modify lines & polygones with graphical method

dbrgn, 2020-05-17:
Nominating for v0.2.0

ubruhin, 2020-05-18:
Similar to https://github.com/LibrePCB/LibrePCB/issues/482.

> Nominating for v0.2.0

I'll add it to the 0.1.5 milestone, althought I'm really not sure whether it is realistic or not (a general refactoring of editors would help to implement this, I already started with the refactoring but it's a lot of work...). Btw, milestone 0.1.5 just means "next minor release" - if the next release is 0.2.0, I'll merge those milestones.

ubruhin, 2020-09-28:
Implemented in #773.

closed 2020-09-28 by ubruhin