LibrePCB/LibrePCB #704: Component search and share : make pain as low as possible

doom-fr, 2020-05-31:
##### SUMMARY
Git is surely a marvelous tool with a bunch of possibilities.
But for electronics guys, I'm not sure this is well known.
So it is a real barrier to share components.

It is also not always easy to know in which library a component or a package is.
It is needed to download the lib to know what's inside.

I think that LibrePCB is really useable at the current stage, but there's lack of librairies.
It is time consuming to create missing components so we have to make libraries contributions as easy as possible, to divide the pain on all users.

##### SOLUTION
For sharing, the git use should be hidden to the user and an HMI for dealing that should be provided.

For searching and downloading components, I think that LibrePCB should provide a tool like [_3D warehouse_](https://3dwarehouse.sketchup.com/) in _sketchup_ (this is the online version but it is also integrated inside Sketkup software)

So to find a component/package, you won't have in your downloaded libraries, just click on a button, enter the name of the component/package, look at all results to see which one is suitable, click to download the component/package (library containing it).

Related to this _3D warehouse_, I think it should be  allowed to share unreviewed components to speed up LibrePCB spreading. Just add tags like 'unreviewed' or 'LibrePCB validated' or a notation, to help user finding the good one. (not sure this will suits to @ubruhin (•‿•) )

I think this is a huge change, but it will help users and so help LibrePCB spreading.

PS : in search tool it should be cool to allow the use of `*` joker character.

ubruhin, 2020-06-01:
Yes, I totally agree. Actually my idea is to allow contributing library improvements through our API, i.e. in addition to downloading libraries through the API (as the library manager does), uploading library elements should be supported too.

Example:
- User creates a new library element within an arbitrary library
- Within the library editor, the user clicks on something like "contribute library element"
- The library element is uploaded to our servers and displayed on some web application
- A maintainer chooses what to do with the contribution, e.g. open a PR on the corresponding Git repository
- Then the normal review procedure is done on that PR

The integration into LibrePCB shouldn't be too hard, but I'm not sure yet what should happen exactly on our server with the uploaded library files (implementing a whole web application requires a lot of effort)...

doom-fr, 2020-06-01:
> The integration into LibrePCB shouldn't be too hard

[image]

doom-fr, 2020-06-02:
And I will have so many things to share !