LibrePCB/LibrePCB #705: Not the good icon in Ubuntu for LibrePCB (Microchip's one instead)

doom-fr, 2020-05-31:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.4
Git Revision:     ae04eef5a71b5ba66ae2cee6b631c1c933ace535
Build Date:       2020-05-02T18:50:04
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.4 LTS
Platform Plugin:  xcb
```

##### SUMMARY
Logo is not the good one
[image]

##### STEPS TO REPRODUCE
1. launch LibrePCB
2. open Microchip library
3. lock screen with super+L
4. log back
Now icon is Microchip one and stay until lock screen again on another window

##### EXPECTED RESULTS
Always LibrePCB icon ?
But at least not Microchip icon when not on Microchip lib window.

##### ACTUAL RESULTS
Icon of Microchip instead of icon of LibrePCB

##### ADDITIONAL INFORMATION
Is it possible to have one icon for each window (not regrouped) ?
It will be easier to switch between all opened ones.
Or maybe a dropdown menu in the icon bar.

ubruhin, 2020-06-01:
Hmm yeah I also noted this strange behavior on Ubuntu some time ago, but I was not able to reproduce it on other operating systems. Thus I really don't know what we are doing wrong :( 

Maybe it's even just an Ubuntu bug?

doom-fr, 2020-06-01:
Maybe a bug in Ubuntu.

What icon should be displayed in theory for each window ?

ubruhin, 2020-06-01:
This is how it looks on Windows:

[image]

Tubbles, 2020-12-27:
I can reproduce this in Manjaro with XFCE, even no need to lock the screen
[image]

ubruhin, 2023-04-15:
I can't reproduce the problem with LibrePCB 0.1.7 on Ubuntu 22.04. Somehow the launcher always shows the LibrePCB logo, no matter which window is opened.

Anyway, generally it's intended that the library icon is used as window icon, to allow distinguishing between multiple opened libraries. How this is represented in a task bar is not really under our control, it's decided by the desktop environment.

So I don't know if there is any action needed on our side. I think it's fine as-is so I'll close this issue. If there are any objections, please reopen :)

closed 2023-04-15 by ubruhin