LibrePCB/LibrePCB #706: Display and offer link to datasheet

doom-fr, 2020-06-01:
##### SUMMARY
Electronics is datasheet, datasheet and datasheet.
But in LibrePCB, there's nothing provided for that.

##### SOLUTION
For each device, we should add datasheet link in the library.
This is spread in the schematic and board editors. Just click to see the datasheet.
For generic component, the field can be manually added in the schematic page.
If not, the user can right click on the field containing the MPN and launch a search in is favorite search engine.

PS : there might be sometimes multiple datasheets for one device. But one is already a great improvement

dbrgn, 2020-06-02:
This would be useful indeed. See also the extended discussion here: https://github.com/LibrePCB/LibrePCB/issues/653 The datasheet could be implemented as a predefined `DATASHEET` attribute.

ubruhin, 2024-11-30:
Implemented in #1460. Not 100% finished yet (UI currently only supports one datasheet per device), but the main functionality is there so I'll close the issue.

closed 2024-11-30 by ubruhin