LibrePCB/LibrePCB #707: Library Editor : Top placement and Documentation colors are too close

doom-fr, 2020-06-02:
##### SUMMARY
Top placement and Documentation are too main layers used when you create a package.
But colors are quite the same, so often bad layer is used.
To fix this, the user should change all bad layer items.
This need sometime to move already placed objects because of https://github.com/LibrePCB/LibrePCB/issues/536

##### SOLUTION
change colors to make a real difference.
And fix #536 :-)

dbrgn, 2020-06-02:
I agree! Do you have a concrete proposal for colors to be used?

doom-fr, 2020-06-02:
Not easy !

Here is a proposition (on left actual, on right new) : 

[image]

dbrgn, 2020-06-03:
That doesn't look too bad actually! @ubruhin what do you think?

@doom-fr what color value did you use here?

dbrgn, 2020-06-04:
Ah, you used image editing... I thought you changed the colors in your config file (`boards/default/settings.user.lp`, see https://github.com/LibrePCB/LibrePCB/issues/406).

You can find the default colors in `libs/librepcb/common/graphics/graphicslayer.cpp`.

doom-fr, 2020-06-06:
With `#76fbc697` for top documentation

[image]

doom-fr, 2020-06-27:
And the problem is mainly in the Library editor and not in the Board editor.
(even if I think both should have same colors)

doom-fr, 2020-06-27:
Maybe it was also a good idea to set top documentation as default layer in Library editor and not top placement, as we start by drawing the component and in second time draw the placement.

closed 2023-10-18 by ubruhin