LibrePCB/LibrePCB #709: Schematic Editor : unconnected pins not showed when multiple (sometimes)

doom-fr, 2020-06-03:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.4
Git Revision:     ae04eef5a71b5ba66ae2cee6b631c1c933ace535
Build Date:       2020-05-02T18:50:04
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.4 LTS
Platform Plugin:  xcb
```

##### SUMMARY
In schematic, little round are at the end of unconnected pins.
Seems they are missing sometimes (observed on multiple pin of same signal)

##### STEPS TO REPRODUCE
1. place an ATmega4809
2. look at GND pins

##### EXPECTED RESULTS
little red round at the end

##### ACTUAL RESULTS
No round

##### ADDITIONAL INFORMATION

VDD is correctly handled

[image]

[image]

[image]

doom-fr, 2020-06-03:
In this component all GND pins should be connected

ubruhin, 2020-06-14:
Maybe it's somehow related to the library component or symbols, could you provide us these library elements somehow?

Usually pins don't show the circle if the corresponding component signal is already connected to a net, e.g. through a different pin which represents the same component signal.