LibrePCB/LibrePCB #710: Board Editor : Remove unnecessary nodes of wires when extending them

doom-fr, 2020-06-05:
##### SUMMARY
When you draw a wire in multiple actions, unnecessary nodes remain there.

[image]

##### SOLUTION
During drawing the first segment of wire that start on another wire of same net and if segments are aligned, remove the start node as it is not useful.
Idem on last segment.

dbrgn, 2020-06-05:
Sometimes you want that intermediate node because you want to mark a specific location. But maybe a "simplify traces" option would be helpful? Or a modifier key, in case you do want the intermediate node to stay?

doom-fr, 2020-06-05:
I think second option is better as usual case is "additional node is not needed".

PR #1591 (2025-08-30): Board editor: Automatically simplify net segments
When drawing traces or deleting traces/vias in the board editor, the modified net segments will be automatically simplified in the following ways:

- Remove redundant junctions (same position, same layer)
- Remove redundant traces (same anchors, same layer), keeping only the thickest of them
- Remove zero-length traces
- Remove useless junctions within straight trace segments (join line segments into the same direction to a single trace)
- Split traces to connect with junctions on the path between start and end points
- Split intersecting trace [... cut]
closed 2025-08-30 by ubruhin