LibrePCB/LibrePCB #711: Board editor : Air wires should go to the closest pad or wire of same net

doom-fr, 2020-06-05:
##### SUMMARY
Seems that air wires goes only to pad or node of wires.
And sometimes, the net is just there very close, but the air wire goes completely in the opposite way because he tries to join the end of the wire and not the wire itself.

[image]

##### SOLUTION

Join the wire directly !

[image]

doom-fr, 2020-06-06:
[image]

[image]

josephduchesne, 2020-10-06:
I've looked through the code a bit, and the basic problem is that AirWiresBuilder is only handed a list of points and edges (two points that are connected). There's no real difference between a point to a wire (what you're after), or a point to a plane (maybe harder to calculate? It might not be?).

My general thought, is:
- After sorting the edge list in AirWiresBuild::kruskaMst(), 

```
foreach ratsnest edge
    foreach edge other than the one we're looping through above
       compute the closest point on that edge (closest point on line segment using some vector math)
       if that point is the start/end point of the edge: continue
       else: add a new point at that closest point, and add a new weight=0 (wire) edge connecting one side of that line to the synthetic point

re-sort the edge list
```
This is a bit brute-force and could be optimized (inserting the synthetic points/edges at the start to avoid the double sort, as well as skipping adding synthetic edges that are strictly worse than both endpoints of another edge, but the result should work as a starting point.

The idea is that every potential ratsnest point will have a synthetic point/edge generated for every edge, and then the Kruskal algorithm can do the heavy lifting. 

This would make the graph go from N= E to N= E+E*A edges, where N is the number of resultant edges, E is the number of all edges, and A is the number of airwire edges.

I'm expecting the code to be fairly minimal at least. Does anyone see a problem with the idea and/or pseudocode?