LibrePCB/LibrePCB #712: Board Editor : Bad net name displayed

doom-fr, 2020-06-06:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.4
Git Revision:     ae04eef5a71b5ba66ae2cee6b631c1c933ace535
Build Date:       2020-05-02T18:50:04
Qt Version:       5.14.2 (built against 5.14.2)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.4 LTS
Platform Plugin:  xcb
```

##### SUMMARY
Net name displayed is not the good one
(But air wiring seems to be good)

##### STEPS TO REPRODUCE
Maybe change net of this pin

##### EXPECTED RESULTS
Show the good name

##### ACTUAL RESULTS
Show the bad name

[image]

##### ADDITIONAL INFORMATION

doom-fr, 2020-06-06:
close & launch the Board Editor does not solve the problem.
close & open the project solve the problem

ubruhin, 2020-06-06:
Yes, I guess when renaming a net, the rename is not propagated to all graphical elements on the board and thus outdated net names are shown.

doom-fr, 2020-06-07:
For information, seems there's quite same problem on via.

[image]

Quite same, but not sure related as I've just placed the via 5s before.

PR #771 (2020-09-16): Fix possibly outdated displayed net names in boards
The displayed net names within vias and pads, and the net names within the hints of vias, pads and traces were not updated properly if a net signal was renamed. So the displayed net names in boards was wrong. Now the event of renaming a net signal is forwarded to vias, pads and traces to enforce updating the displayed texts and hints accordingly.

Fixes #712
closed 2020-09-16 by ubruhin