LibrePCB/LibrePCB #718: Failed functional tests on macOS with Qt 5.15

ubruhin, 2020-06-21:
##### VERSION / OS / ENVIRONMENT

Current master branch on macOS CI environment using Qt 5.15.

##### SUMMARY

In #716 I had to mark some functional tests as allowed to fail since they suddenly failed on macOS and I have no idea why.

##### STEPS TO REPRODUCE

Run the functional tests.

##### EXPECTED RESULTS

All tests pass.

##### ACTUAL RESULTS

Following tests fail:

```
tests/funq/librarymanager/test_create_local_library.py::test FAILED      [ 56%]
tests/funq/librarymanager/test_download_manually.py::test FAILED         [ 59%]
tests/funq/librarymanager/test_install_remote_libraries.py::test FAILED  [ 62%]
```

According the console output, it looks like added/created libraries do not appear in the library list of the library manager.

##### ADDITIONAL INFORMATION

They worked with Qt 5.14, so it looks like it's related to the Qt version. But I couldn't reproduce it on Linux and Windows, and since I don't have macOS I'm not able to test it on macOS. Actually I don't even know whether only the tests fail, or if there is indeed something wrong with LibrePCB.

hugs, 2020-07-16:
Hi!

I looked into these test failures and I was able to reproduce the issue on MacOS Catalina (10.15.5) and Qt 5.15.0. I think I found the exact spot where the tests are failing. They're all related to clicking a button. More importantly, I think they uncovered an actual bug in the UI! In the actual UI (as in the tests), the user needs to click a button twice to get the action to work. (It's not just a quirk of the test or test framework. I'll look into creating a new issue for this bug. It'll probably be titled something like "Buttons in Library Manager require two clicks to work.")

**In [test_create_local_library.py](https://github.com/LibrePCB/LibrePCB/blob/9d06ac2c96e5ad7f911c7de92d571778399dcb42/tests/funq/librarymanager/test_create_local_library.py#L42)**
The test fails here:

    app.widget('libraryManagerCreateLocalLibraryCreateButton').click()

However, the test passes if you add an `"activate_focus()"` right before the click like this:

    app.widget('libraryManagerCreateLocalLibraryCreateButton').activate_focus()
    app.widget('libraryManagerCreateLocalLibraryCreateButton').click()

It *also* passes if you add a second click, like so:

    app.widget('libraryManagerCreateLocalLibraryCreateButton').click()
    app.widget('libraryManagerCreateLocalLibraryCreateButton').click()

**In [test_download_manually.py](https://github.com/LibrePCB/LibrePCB/blob/9d06ac2c96e5ad7f911c7de92d571778399dcb42/tests/funq/librarymanager/test_download_manually.py#L43):**

The test fails here:

    app.widget('libraryManagerDownloadManuallyDownloadButton').click()

However, same as with test_create_local_library.py, adding an `activate_focus()` fixes the issue:

    app.widget('libraryManagerDownloadManuallyDownloadButton').activate_focus()
    app.widget('libraryManagerDownloadManuallyDownloadButton').click()

This also works:

    app.widget('libraryManagerDownloadManuallyDownloadButton').click()
    app.widget('libraryManagerDownloadManuallyDownloadButton').click()
    
    
**In [test_install_remote_libraries.py](https://github.com/LibrePCB/LibrePCB/blob/9d06ac2c96e5ad7f911c7de92d571778399dcb42/tests/funq/librarymanager/test_install_remote_libraries.py#L63):**

The test fails here:

    app.widget('libraryManagerDownloadFromRepoDownloadButton').click()

However, same as with test_create_local_library.py, adding an `activate_focus()` fixes the issue:

    app.widget('libraryManagerDownloadFromRepoDownloadButton').activate_focus()
    app.widget('libraryManagerDownloadFromRepoDownloadButton').click()

This also works:

    app.widget('libraryManagerDownloadFromRepoDownloadButton').click()
    app.widget('libraryManagerDownloadFromRepoDownloadButton').click()  
    
    
Here's a video of test_create_local_library.py with and without the fix.
[image]

Here's a video of the actual bug in the user interface showing that two clicks are required to get the "Create Library" button to work. (Test run manually.)
[image]

dbrgn, 2020-07-16:
Awesome work! Could this be a Qt bug, @ubruhin?

hugs, 2020-07-16:
My working theory (from what I've found so far online) is that there is a missing "activateWindow()" call somewhere in the UI setup code. Basically, the two clicks are currently required because QPushButton don't respond to clicks **unless the window is active.** That first click effectively activates the window, and the second click now "works as expected" because the window is now active. Which also explains why adding "activate_focus()" to the test makes it pass without an additional "click()".

ubruhin, 2020-07-18:
Very interesting, thanks for the investigation @hugs! :tada: 

Somehow this sounds like a Qt bug, but so far I didn't find such a bug report :( I mean, it worked properly for several years, and still works properly on other operating systems, why should the behavior suddenly change on macOS?

Btw, actually we even explicitly activate the window after opening it:

https://github.com/LibrePCB/LibrePCB/blob/9d06ac2c96e5ad7f911c7de92d571778399dcb42/apps/librepcb/controlpanel/controlpanel.cpp#L506

hugs, 2020-07-18:
Happy to help! Does `activateWindow()` have to be called again when changing between the 3 different tabs of the library manager? (Sorry, I'm still new to Qt, but I'm trying to catch up quick!)

ubruhin, 2020-07-18:
> Does activateWindow() have to be called again when changing between the 3 different tabs of the library manager?

No, actually I guess calling that function is almost never needed. In this situation we call it only for the case the library manager is already opened in the background, to bring it to foreground instead of opening a new instance. In general, managing the focus / active window is handled by Qt and/or the window manager.

hugs, 2020-07-19:
Maybe this is a Qt bug afterall. :-) I did find this issue when I searched for "QPushButton 5.15 macOS":
[QPushButton with stylesheet cannot be pushed when it is clicked on macOS & Qt 5.15](https://bugreports.qt.io/browse/QTBUG-84879)

There's even a patch that has been merged just a few days ago that we could test.

ubruhin, 2020-09-17:
Looks like it was a Qt bug indeed - now with Qt 5.15.1 the tests are successful again! :rocket: :smiley: 

So probably with Qt 5.15.0 LibrePCB does not work properly, but I think it's not worth to implement a read workaround for this specific bug. Let's just hope nobody uses LibrePCB with Qt 5.15.0 on macOS :wink: Our official binary releases will contain the latest available Qt version anyway, so they should work fine.

Reverted the workaround in #772

PR #772 (2020-09-19): Functional tests: Revert marking failed tests on macOS with xfail
This reverts commit feea5c1ef115d83a8090132f7fbdf40c1a41db65 which was added in #716. It seems that the problem was "automatically" solved when macOS updated Qt to 5.15.1 :wink:

Fixes #718
closed 2020-09-19 by ubruhin