LibrePCB/LibrePCB #726: Maximize working area by panels with auto-hide and pinning possibility

EdizonTN, 2020-07-10:
The work in the working area is a primary task for each designer.

Due to this, I propose all toolboxes and panels (Icon toolbar, Layers panel, ERC panel, Command, etc.. ) be rebuilt as "**panels with auto hide feature**" with "_Pinning_" possibility. Of course, settings are saved into the workspace setting file.
It's important for SCH and PCB editors.
The library should be applied only for the symbol editor and the package editor.

On the GitHub is free code for this: https://github.com/anyox/Qt-Auto-Hide-DockWidget

Better would be if the show hidden panel will be activated by "mouse over" event :-) opposite of click (minimize clicks).

Thanks!

dbrgn, 2020-12-27:
Hm, that project you linked does not look very maintained (it appears that there isn't even any documentation). It would probably mean we'd have to maintain the library ourselves (and port it to future Qt versions).

Maybe there are other solutions?

EdizonTN, 2020-12-27:
I'm not a Qt programmer, but I found another tool: https://github.com/KDAB/KDDockWidgets
I don't know if will be good....

dbrgn, 2020-12-27:
Oh, that looks *much* better! It's also offered as a commercial offering, so there will be continued development and support.

https://www.kdab.com/development-resources/qt-tools/kddockwidgets/

@ubruhin what do you think about dockable panes in general?

ubruhin, 2025-08-04:
The new UI (see #1494) introduces a very new layout of the editors, and increases the working area in several ways. Therefore maybe this issue is not relevant anymore. Let's first gain some experience with the new UI of LibrePCB 2.0, and if there is still a need to hide the sidebar or so, let's address it in a new issue.

closed 2025-08-04 by ubruhin