LibrePCB/LibrePCB #728: Board Editor: Move and rotate unaligns devices

5n8ke, 2020-07-10:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.4
Git Revision:     ae04eef5a71b5ba66ae2cee6b631c1c933ace535
Build Date:       2020-05-02T18:50:43
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Windows 10 (10.0)
Platform Plugin:  windows
```

##### SUMMARY
When a device, aligned to the snap grid, is selected, moved with left hold, and then rotated with right-click, the alignment is destroyed.

##### STEPS TO REPRODUCE
Open a board with a device aligned to the snap grid, select with left click, move with left hold, and rotate 1-3 times with right-click.

##### EXPECTED RESULTS
An aligned device is still aligned to the snap grid.

##### ACTUAL RESULTS
The device is no longer aligned.

##### ADDITIONAL INFORMATION
- It seems, that the rotation origin does not match the displayed device origin.
- The offset rotation origin seems to change with the relation of the actual mouse pointer in regard to the device origin. A bigger distance also results in a bigger offset of the origin.
- After four rotations, the device is again aligned with the snap grid.

doom-fr, 2020-07-10:
Right. I have reproduced it.
Same with rotation using **'r'** key

doom-fr, 2021-02-23:
I think this is related : the text moves when rotating

[image]

ubruhin, 2023-01-14:
The original issue should be fixed with #1082 (at least I could not reproduce it anymore). The problem of the moving items after rotation is not fixed yet, but there's already the issue #450 open for that.

PR #1082 (2023-01-14): Fix possibly unexpected center of rotate/mirror operations
When rotating/mirroring items in any of the editors, so far the center point of the operation was always mapped to the grid interval. However, this is not expected when only a single item is selected since it moves the item's position if it was not aligned to the grid. Thus now using the item's exact center position (not mapped to grid) if only one is selected. For example rotating a single, unaligned pad now keeps its exact position.

If more than one item is selected, the behavior is not affected by this change.

This also seems to f [... cut]
closed 2023-01-14 by ubruhin