LibrePCB/LibrePCB #734: The workspace cannot be opened after library edit or crash

miniwark, 2020-07-20:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.4 from Flathub
Git Revision:
Qt Version:
Operating System: Linux Manjaro (Arch)
...
```

##### SUMMARY

When starting the application i get the following message, even after a reboot:

> The workspace "/home/miniwark/Documents/Developpements/PCB/LibrePCB" cannot be opened: The workspace is already opened by another application instance or user!

The LibrePCB just quit. (may be related to #154)

##### STEPS TO REPRODUCE

It had happen randomly times already. I think it's related to crash and/or bad saves during local Library editing (but a also have the case without crashes).

I suspect than there is a temporary file somewhere (instead of a PID) who keep track of open library parts, but who is not correctly deleted.

##### EXPECTED RESULTS

LibrePCB must offer me a way to try to force my way after the warning instead of just quitting.

miniwark, 2020-07-20:
(as a workaround can you tel me witch file to delete instead of recreate / re-downloard libraries ?)

ubruhin, 2020-07-20:
Actually the lock should be ignored if the process which created the lock is no longer running. This might not work if for example the PID could not be determined. Could you post the content of your `LibrePCB-Workspace/v0.1/.lock` file (feel free to obfuscate username etc.)?

miniwark, 2020-07-21:
@ubruhin, sure:

```
Username
username
mycomputer
2
librepcb
2020-07-19T09:18:55Z
```

ubruhin, 2020-07-21:
Thanks! It looks good, but the PID `2` looks strange, which helped me to find the actual problem: It seems that Flatpak uses PID namespacing, thus LibrePCB probably still has PID=2 after a crash, therefore the file lock cannot be considered as stale. See also https://github.com/flatpak/flatpak/issues/1963. I'm not sure if we can do anything to fix this, I suspect detecting stale file locks in a reliable way is impossible if PIDs are namespaced :(

> LibrePCB must offer me a way to try to force my way after the warning instead of just quitting.

Yeah, probably it's the only workaround to provide a way to ignore the file lock, with the risk that users ignore file locks by accident which possibly leads to much more serious issues like corrupt projects and workspaces. I'm not sure if we should provide such a dangerous option... Or at least only provide it if the PID is "suspect" (<10 or so).

miniwark, 2020-08-01:
Or maybe use the file timestamp for the "force option" ?

kilt-er, 2021-02-23:
I am having this same problem.  If I delete the PID line, then I can start the program.  However, when I go to open my project, I get:

"The directory is locked, check if it is already opened elsewhere: [project]"

What should I do?

kilt-er, 2021-02-23:
I managed to find two more .lock files.  I did the sam PID clear, and I was able to get myself unblocked after an hour-and-a-half of freaking out.

Wondering if there is anything that can be done here to more accurately determine when a lock has been involved in a crash and clear it.

miniwark, 2021-02-23:
@kilt-er in my case, for a workaround, i have simply deleted the `.lock` file in the library instead of editing it (because i was sure than i have only one LibrePCB running).

kilt-er, 2021-02-24:
> @kilt-er in my case, for a workaround, i have simply deleted the `.lock` file in the library instead of editing it (because i was sure than i have only one LibrePCB running).

Thanks for the info.  I tried that and it led to a different error message about the lock file being missing.  The lock is held after reboot -- no multiple instance concerns.  It seems like legit bug that the program would fail to recognize a lock held in error after crash when only one instance is running.

ubruhin, 2021-02-24:
@kilt-er Did you install LibrePCB with Flatpak? If not, which operating system are you using?

kilt-er, 2021-02-24:
Ah, sorry for leaving that out.  Yes, I'm using Flatpak on the latest Rasbian OS for Rpi 4

ubruhin, 2021-02-26:
OK so the main problem is that the same PID gets reused because Flatpak uses PID namespacing, and thus LibrePCB thinks the application holding the lock is still running.

PR #871 (2021-03-04): Allow opening locked workspace/project/library in some cases
If a workspace, project or library directory is locked by a `.lock` file but we are not 100% sure if the lock is still valid, the error message dialog now provides the ability to discard the lock and open the directory anyway:

[image]

This is especially useful when running LibrePCB as a Flatpak since due to PID namespacing we are not able to check if the application holding a lock is still running or not.

But if we are absolutely sure the lock is still valid, this option is not available since generally it could be dangerous to di [... cut]
closed 2021-03-04 by ubruhin